rm-hull/luma.oled #166: How to change default display of your examples (luma.oled)?

rezaaghazade, 2017-09-15:
hi body.
i want to use your luma.oled example, 
but i have a problem
when i run the code, display bellow 👍 

_Display : pcd8544
Interface : i2c
Dimensions : 128*64_

-------
my display is not pcd8544
how can i change the display to what i use ?

attention : my oled worked with AdaFruit Code, but not work with your code 
https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code

rm-hull, 2017-09-15:
Hi, so what type of device have you got, and do you know its model number? If you aren't sure, please take a photo (of the front and the back of the display) and drag it into your reply.
 
and what is the python command you are running? I am assuming you are running one of the examples, but I need to know which one and what flags you are supplying.

rezaaghazade, 2017-09-16:
this is my oled, i think this is ssd1306 128*64 (2Color)
[image]
[image]
-------------------
and about a command i use : 
i use simple command 👍 
_python yourExampleName.py_
this code work for another example except yours

rm-hull, 2017-09-16:
Ok, so it looks like a regular 128x64 SSD1306 using I2C - this is probably the most common device and is fully supported.

Check to make sure you have installed the **luma.oled** package from pypi:

    $ pip list | grep luma 
    luma.core (1.0.1)
    luma.emulator (0.2.5)
    luma.lcd (1.0.0)
    luma.led-matrix (1.0.2)
    luma.oled (2.2.10)

Your list might be different, but you need to ensure **luma.oled** is installed - if you can't see it, try installing again with:

    $ sudo pip install -IU luma.oled

Clone the http://github.com/rm-hull/luma.examples repo if you haven't done already, and from within that directory (any of the examples should work the same):

    $ python examples/clock.py -d ssd1306

Look at the source code and the docs for how to use the classes in your own program

rezaaghazade, 2017-09-17:
thanks alot,
it's worked
-------------------------
after all these, i have another question,
my oled is 2 color,
2 color means every pixel can show 2 color ?
how can i change these color (with your code)?

rm-hull, 2017-09-17:
I have a device exactly the same as yours - it has a yellow band at the top and and the remainder of the screen is blue, and i originally thought that each pixel color could be programmed accordingly but it is not possible - the SSD1306 is monochrome, presumably there is a filter of some sort built into the display.

There are other OLED displays which support full color, eg the [SSD1331](https://www.aliexpress.com/item/Free-Shipping-0-95-inch-7pin-Full-Color-65K-Color-ssd1331-OLED-Display-3-3v-5v/32646204260.html) - also supported by this library, and works in full color.

closed 2017-09-19 by thijstriemstra