rm-hull/luma.oled #171: module not found

Pedotech, 2017-10-25:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
Raspbian

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.
tried several installations always get message luma.device not found.
Actually it is in egg
-Sorry I am new to this

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

thijstriemstra, 2017-10-26:
@Pedotech have you followed these instructions? https://luma-oled.readthedocs.io/en/latest/install.html

And what version of Raspbian are you using?

Pedotech, 2017-10-28:
Thank you so much for all your generous work!
I reinstalled everything on a new sd and it is working fine.
I have an additional oled on SPI. How can I use it? Message is "oled not found at I2C!
Thank you againa

rm-hull, 2017-10-28:
> I have an additional oled on SPI. How can I use it? Message is "oled not found at I2C!

First, look at the initial example here: https://luma-oled.readthedocs.io/en/latest/python-usage.html. Also, have a read of the [hardware](https://luma-oled.readthedocs.io/en/latest/hardware.html) section.

Try one of the [examples](https://github.com/rm-hull/luma.examples), as follows:

    $ python examples/demo.py --display=ssd1306 --interface=spi

There are a few other config options, but the defaults are sensible and should work out-of-the-box.

closed 2017-10-31 by thijstriemstra