rm-hull/luma.oled #172: RPi3 Install Issue

jethornton, 2017-10-26:
When I followed the install from the user directory it failed, during trouble shooing I changed to 
pi@raspberrypi:/usr/local/lib/python2.7/dist-packages $ 
and did the install again and it worked.

[install.txt](https://github.com/rm-hull/luma.oled/files/1418187/install.txt)

thijstriemstra, 2017-10-26:
This looks like a pip issue, see https://stackoverflow.com/questions/37495375/python-pip-install-throws-typeerror-unsupported-operand-types-for-retry

When you follow the instructions at https://luma-oled.readthedocs.io/en/latest/install.html it mentions to upgrade setuptools and pip first, did you follow these steps?

thijstriemstra, 2017-10-31:
@jethornton any news?

closed 2017-11-20 by thijstriemstra

CoWinkKeyDinkInc, 2017-11-27:
I'm still having trouble with this problem.  When I try to type `apt-get remove python-pip python3-pip` then it says permission denied, after I use `sudo` with the command to force it, it says this:
```
pi@raspberrypi:~ $ sudo apt-get remove python-pip python3-pip
E: Could not get lock /var/lib/dpkg/lock - open (11: Resource temporarily unavailable)
E: Unable to lock the administration directory (/var/lib/dpkg/), is another process using it?
```

rm-hull, 2017-11-27:
@CoWinkKeyDinkInc - when I see that, I usually assume that the system is being automatically updated in the background, so I back off and try later - usually this works fine.

If you are getting it the whole time, have a look at https://askubuntu.com/questions/15433/unable-to-lock-the-administration-directory-var-lib-dpkg-is-another-process