rm-hull/luma.oled #174: Add support for SSD1351 128x128 Color OLED

rm-hull, 2017-10-27:
_[this is a placeholder to record progress on implementation of the `ssd1351` device]_

Code will be committed on the **feature/ssd1351** branch prior to merging into master. 

This development was permitted by the kind donation of funds by @svenyonson to purchase https://www.aliexpress.com/item/1-5-inch-Colorful-OLED-Module-SSD1331-128x128-Resolution-for-51-STM32-Arduino/32721463778.html

Item received, close up pics as follows:
[image]

[image]

Datasheet: http://www.hpinfotech.ro/SSD1351.pdf

rm-hull, 2017-10-28:
Using @notro's fbtft driver (swapping over RES & DC from prev comment) confirms that the display works OK:

    $ sudo modprobe fbtft_device name=freetronicsoled128
    $ dmesg | grep fbtft_device
    [   63.900658] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
    [   63.903323] fbtft_device: GPIOS used by 'freetronicsoled128':
    [   63.903337] fbtft_device: 'reset' = GPIO24
    [   63.903345] fbtft_device: 'dc' = GPIO25
    $ conf2fbmap 1 1

[image]

rm-hull, 2017-10-28:
Ok, so we now have a working implementation:

[image]

[image]

Performance on a Raspberry Pi B2:

```
$ python examples/perfloop.py -f conf/ssd1351.conf --framebuffer=full_frame
Testing display rendering performance
Press Ctrl-C to abort test

Display: ssd1351
Interface: spi
Dimensions: 128 x 128
----------------------------------------
#### iter =    651: render time = 60.09 ms, frame rate = 16.61 FPS
```

rm-hull, 2017-10-30:
Fixed in #177

closed 2017-10-30 by rm-hull

martijnrondeel, 2017-11-19:
A few months ago I was searching for a proper library and came across this one but it didn't have support back then. Now it does, thank you so much for supporting SSD1351 👍.

nanomesher, 2018-04-12:
I have exactly the same display as above

**Tried both connections as below:**

OLED Pin | OLED Name | RPi Pin | RPi Function | Remarks
-- | -- | -- | -- | --
1 | GND | P01-20 | GND | Ground
2 | VCC | P01-17 | 3V3 | +3.3V Power
3 | SCL | P01-23 | GPIO 11 (SCLK) | Serial Clock
4 | SDA | P01-19 | GPIO 10 (MOSI) | Serial Data
5 | RES | P01-22 | GPIO 25 | Reset
6 | DC | P01-18 | GPIO 24 | Data/Command
7 | CS | P01-24 | GPIO 8 (CE0) | Chip Select

OLED Pin | OLED Name | RPi Pin | RPi Function | Remarks
-- | -- | -- | -- | --
1 | GND | P01-20 | GND | Ground
2 | VCC | P01-17 | 3V3 | +3.3V Power
3 | SCL | P01-23 | GPIO 11 (SCLK) | Serial Clock
4 | SDA | P01-19 | GPIO 10 (MOSI) | Serial Data
5 | RES | P01-18 | GPIO 24 | Data/Command
6 | DC | P01-22 | GPIO 25 | Reset
7 | CS | P01-24 | GPIO 8 (CE0) | Chip Select

Checked many times that spi port seems ok

 ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Apr 12 15:40 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr 12 15:40 /dev/spidev0.1

The program runs without error, but I see nothing on the display , any ideas?

:~/luma.examples$ python examples/perfloop.py -f conf/ssd1351.conf --framebuffer=full_frame
Testing display rendering performance
Press Ctrl-C to abort test

Version: luma.oled 2.3.2 (luma.core 1.7.2)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------
#### iter =   2945: render time = 43.67 ms, frame rate = 22.87 FPS

Thanks in advance!

ayhlai, 2018-04-12:
I have exactly the same display as above

**Tried both connections as below:**

OLED Pin | OLED Name | RPi Pin | RPi Function | Remarks
-- | -- | -- | -- | --
1 | GND | P01-20 | GND | Ground
2 | VCC | P01-17 | 3V3 | +3.3V Power
3 | SCL | P01-23 | GPIO 11 (SCLK) | Serial Clock
4 | SDA | P01-19 | GPIO 10 (MOSI) | Serial Data
5 | RES | P01-22 | GPIO 25 | Reset
6 | DC | P01-18 | GPIO 24 | Data/Command
7 | CS | P01-24 | GPIO 8 (CE0) | Chip Select

OLED Pin | OLED Name | RPi Pin | RPi Function | Remarks
-- | -- | -- | -- | --
1 | GND | P01-20 | GND | Ground
2 | VCC | P01-17 | 3V3 | +3.3V Power
3 | SCL | P01-23 | GPIO 11 (SCLK) | Serial Clock
4 | SDA | P01-19 | GPIO 10 (MOSI) | Serial Data
5 | RES | P01-18 | GPIO 24 | Data/Command
6 | DC | P01-22 | GPIO 25 | Reset
7 | CS | P01-24 | GPIO 8 (CE0) | Chip Select

**Checked many times that spi port seems ok**

 ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Apr 12 15:40 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr 12 15:40 /dev/spidev0.1

**The program runs without error, but I see nothing on the display , any ideas?**

:~/luma.examples$ python examples/perfloop.py -f conf/ssd1351.conf --framebuffer=full_frame
Testing display rendering performance
Press Ctrl-C to abort test

Version: luma.oled 2.3.2 (luma.core 1.7.2)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------
#### iter =   2945: render time = 43.67 ms, frame rate = 22.87 FPS

Thanks in advance!

rm-hull, 2018-04-12:
* Double check your wiring
* Paste close-up pics of the front and back of the oled display board as a comment here

ayhlai, 2018-04-14:
Thanks for your reply.  I have double checked my wiring again and it looks correct.
I have also tried 
- a brand new Raspbian installation
- on both Pi 3 and Pi 3 B+.
- 2 displays I have , both using SSD1351

Wiring

OLED Pin | OLED Name | RPi Pin | RPi Function | Remarks
-- | -- | -- | -- | --
1 | GND | P01-20 | GND | Ground
2 | VCC | P01-17 | 3V3 | +3.3V Power
3 | SCL | P01-23 | GPIO 11 (SCLK) | Serial Clock
4 | SDA | P01-19 | GPIO 10 (MOSI) | Serial Data
5 | RES | P01-18 | GPIO 24 | Reset
6 | DC | P01-22 | GPIO 25 | Data/Command
7 | CS | P01-24 | GPIO 8 (CE0) | Chip Select

All yield same results - no errors, nothing on screen.  It happens on both of my SSD1351.  
Also,  ssd1306 i2c display works without issue.

Here are some photos (both display were tested):

[image]

[image]

Any suggestions of how to debug? thanks very much.

thijstriemstra, 2018-04-14:
@ayhlai can you open a new ticket for this? thanks for the detailed bug reports.

rm-hull, 2018-04-15:
@thijstriemstra good idea

@ayhlai please create a new issue and we'll try to help you. When you do create it, please also include the command line you used to start the python script and any output it may have shown. If it was one of your own programs, please include that too. Thanks.

thijstriemstra, 2018-04-18:
> @thijstriemstra good idea

Can you give me these github permissions, seems I'm not able to lock issues right now.

rm-hull, 2018-04-18:
@thijstriemstra do you know where/what setting I need to change? I can't spot anything obvious.

thijstriemstra, 2018-04-18:
@rm-hull at the bottom

[image]