rm-hull/luma.oled #175: Installation problem on RPi3

Jaeger87, 2017-10-27:
RPi3

#### Linux Kernel version
Linux raspberrypi 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux

Hi, i tried to install the library by following the guide on: https://luma-oled.readthedocs.io/en/latest/install.html

I tried by simply copying all the lines but installation fails cause sudo: pip: is not more a recognized command.

I tried the installation without the third line (the one that purge pip) but than i got this error:

`Collecting luma.oled
  Downloading luma.oled-2.2.12-py2.py3-none-any.whl
Collecting luma.core>=1.0.3 (from luma.oled)
  Downloading luma.core-1.0.3-py2.py3-none-any.whl (49kB)
    100% |################################| 51kB 737kB/s
Collecting pillow>=4.0.0 (from luma.core>=1.0.3->luma.oled)
  Using cached Pillow-4.3.0.tar.gz
Collecting monotonic; python_version < "3.3" (from luma.core>=1.0.3->luma.oled)
  Downloading monotonic-1.3-py2.py3-none-any.whl
Requirement already up-to-date: smbus2 in /usr/local/lib/python2.7/dist-packages (from luma.core>=1.0.3->luma.oled)
Requirement already up-to-date: RPI.GPIO; platform_system == "Linux" in /usr/lib/python2.7/dist-packages (from luma.core>=1.0.3->luma.oled)
Collecting spidev; platform_system == "Linux" (from luma.core>=1.0.3->luma.oled)
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 809, in unpack_url
    unpack_file_url(link, location, download_dir, hashes=hashes)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 715, in unpack_file_url
    unpack_file(from_path, location, content_type, link)
  File "/usr/lib/python2.7/dist-packages/pip/utils/__init__.py", line 617, in unpack_file
    flatten=not filename.endswith('.whl')
  File "/usr/lib/python2.7/dist-packages/pip/utils/__init__.py", line 502, in unzip_file
    zip = zipfile.ZipFile(zipfp, allowZip64=True)
  File "/usr/lib/python2.7/zipfile.py", line 770, in __init__
    self._RealGetContents()
  File "/usr/lib/python2.7/zipfile.py", line 813, in _RealGetContents
    raise BadZipfile, "File is not a zip file"
BadZipfile: File is not a zip file
`

Thank you!

rm-hull, 2017-10-28:
Can you confirm the version of `pip` you have installed please, as per below:

    $ pip --version
    pip 9.0.1 from /usr/local/lib/python2.7/dist-packages (python 2.7)

If it isn't 9.0.1, then this is _possibly_ what the issue is. Make sure pip is upgraded by running:

    $ sudo -i pip install --upgrade pip setuptools

And then check the version again before proceeding.

----
Alternatively, if you already are on pip 9.0.1, then it could be that a file got corrupted when you downloaded it, so try:

    $ sudo pip install -IU --no-cache-dir luma.oled

Jaeger87, 2017-10-28:
Pip was already updated but it seeems that the file was corrupted like you seggested. However i got a new error when the setup.py for pillow is running:

`sudo pip install -IU --no-cache-dir luma.oled
Collecting luma.oled
  Downloading luma.oled-2.2.12-py2.py3-none-any.whl
Collecting luma.core>=1.0.3 (from luma.oled)
  Downloading luma.core-1.1.0-py2.py3-none-any.whl (51kB)
    100% |################################| 51kB 1.1MB/s
Collecting pillow>=4.0.0 (from luma.core>=1.0.3->luma.oled)
  Downloading Pillow-4.3.0.tar.gz (13.9MB)
    100% |################################| 13.9MB 3.4MB/s
Collecting monotonic; python_version < "3.3" (from luma.core>=1.0.3->luma.oled)
  Downloading monotonic-1.3-py2.py3-none-any.whl
Collecting smbus2 (from luma.core>=1.0.3->luma.oled)
  Downloading smbus2-0.2.0.tar.gz
Collecting RPI.GPIO; platform_system == "Linux" (from luma.core>=1.0.3->luma.oled)
  Downloading RPi.GPIO-0.6.3.tar.gz
Collecting spidev; platform_system == "Linux" (from luma.core>=1.0.3->luma.oled)
  Downloading spidev-3.2.tar.gz
Collecting olefile (from pillow>=4.0.0->luma.core>=1.0.3->luma.oled)
  Downloading olefile-0.44.zip (74kB)
    100% |################################| 81kB 8.3MB/s
Installing collected packages: olefile, pillow, monotonic, smbus2, RPI.GPIO, spidev, luma.core, luma.oled
  Running setup.py install for olefile ... done
  Running setup.py install for pillow ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ujgjAV/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-XZn2ZM-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-2.7
    creating build/lib.linux-armv7l-2.7/PIL
    copying PIL/PaletteFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ExifTags.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/SpiderImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageQt.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/Jpeg2KImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/PpmImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/SunImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/GimpPaletteFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageMode.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/PcdImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImagePath.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageStat.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/BmpImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/SgiImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ContainerIO.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/CurImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/PdfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/PSDraw.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/Image.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/FpxImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageGrab.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/TiffImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageChops.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/WalImageFile.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageEnhance.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageSequence.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/WmfImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageShow.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/Hdf5StubImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/__init__.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/ImageTransform.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/EpsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/FliImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/IptcImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/features.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/IcoImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
    copying PIL/IcnsImagePlugin.py -> build/lib.linux-armv7l-2.7/PIL
[... 144 lines cut ...]
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-9xgeTe/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DHAVE_LIBJPEG -DHAVE_LIBZ -DPILLOW_VERSION="4.3.0" -I/usr/include/freetype2 -I/tmp/pip-build-ujgjAV/pillow/libImaging -I/usr/include -I/usr/local/include -I/usr/include/python2.7 -I/usr/include/arm-linux-gnueabihf -c libImaging/BoxBlur.c -o build/temp.linux-armv7l-2.7/libImaging/BoxBlur.o
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-9xgeTe/python2.7-2.7.13=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DHAVE_LIBJPEG -DHAVE_LIBZ -DPILLOW_VERSION="4.3.0" -I/usr/include/freetype2 -I/tmp/pip-build-ujgjAV/pillow/libImaging -I/usr/include -I/usr/local/include -I/usr/include/python2.7 -I/usr/incl [... cut ...]
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ujgjAV/pillow/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-XZn2ZM-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-ujgjAV/pillow/
`

In this thread it seems that someone got the same error and he was a dependency issue: https://github.com/python-pillow/Pillow/issues/2163

I don't know what dependencies miss in my case, any suggestion?

rm-hull, 2017-10-28:
* What model of Raspberry Pi are you using ?
* Which distribution of Raspbian are you using ? Jessie or Stretch?
* Do you have X-windows running on the Pi ?
* How much free memory and swap used does the `free` command report ?
* Are you using a high quality micro-SD card ?

I think the linked issue is possibly not the root cause. Looking at the output, 

```
libImaging/Histo.c: In function 'ImagingGetHistogram':
libImaging/Histo.c:169:1: internal compiler error: Errore di segmentazione
}
^
Please submit a full bug report,
with preprocessed source if appropriate.
See file:///usr/share/doc/gcc-6/README.Bugs for instructions.
```
I am assuming that _"Errore di segmentazione"_ is a segmentation fault that occurred in the compiler. I would try updating your RPi to the latest set of packages (apt update), power it off and on again, and then try installing pillow first:

    $ sudo pip install -IU --no-cache-dir pillow

If that works, then (noting the omitted flags):

    $ sudo pip install -U luma.oled

thijstriemstra, 2017-10-30:
If it ends with:

```
arm-linux-gnueabihf-gcc: error: build/temp.linux-armv7l-2.7/libImaging/JpegDecode.o: File o directory non esistente
```

the jpeg library is missing. And it's a Pillow requirement so I'm contemplating adding the Pillow dependencies to the installation docs as well.

Anyway, @Jaeger87 can you follow these instructions first and then try luma.oled installation again: http://pillow.readthedocs.io/en/4.3.x/installation.html#building-on-linux

Jaeger87, 2017-10-30:
It seems that the solution from rm-hull worked, i run the command:

`sudo pip install -IU --no-cache-dir pillow`

and than:

`sudo pip install -IU --no-cache-dir luma.oled`

and finally

`Successfully installed RPI.GPIO-0.6.3 luma.core-1.1.0 luma.oled-2.3.0 monotonic-1.3 olefile-0.44 pillow-4.3.0 smbus2-0.2.0 spidev-3.2
`

I tried one example with emulator for test but it didn't start :(

This is the command:

`python clock.py --display pygame
`

And this is the respond

`usage: clock.py [-h] [--config CONFIG] [--display] [--width WIDTH]
                [--height HEIGHT] [--rotate] [--interface]
                [--i2c-port I2C_PORT] [--i2c-address I2C_ADDRESS]
                [--spi-port SPI_PORT] [--spi-device SPI_DEVICE]
                [--spi-bus-speed SPI_BUS_SPEED] [--gpio GPIO]
                [--gpio-data-command GPIO_DATA_COMMAND]
                [--gpio-reset GPIO_RESET] [--gpio-backlight GPIO_BACKLIGHT]
                [--block-orientation] [--mode] [--framebuffer] [--bgr]
                [--h-offset H_OFFSET] [--v-offset V_OFFSET]
                [--backlight-active]
clock.py: error: argument --display/-d: invalid choice: 'pygame' (choose from 'ssd1306', 'ssd1322', 'ssd1325', 'ssd1331', 'ssd1351', 'sh1106')
`

It can be cause by the installation?

thijstriemstra, 2017-10-30:
did you install the `luma.emulator` package?

```
sudo pip install -IU --no-cache-dir luma.emulator
```

Jaeger87, 2017-10-30:
It worked!

Thank you rm-hull and thijstriemstra, you are very kind and professional, i will put yours name and the library link in my blog when i finish this project, thank you again!

You can close this thread!

rm-hull, 2017-10-30:
@Jaeger87 good stuff! Drop a link to your blog in the comments 👍

closed 2017-10-30 by rm-hull

Jaeger87, 2017-10-30:
www.jaegerbox.net
The project is a smartchalkboard controllable by telegram, i hope to finish it in this month and publish code and other stuff in early 2018.