rm-hull/luma.oled #178: baseline_data.py is 236 KB

thijstriemstra, 2017-10-31:
https://github.com/rm-hull/luma.oled/blob/master/tests/baseline_data.py is currently 236 KB and filled with data that is not appropriate to store in a python source file. I think we should move it to separate files, something like `.dat` or whatever. It could hold a pickled version of the data which would be loaded for every test (suite).

rm-hull, 2017-10-31:
Yup it's getting (gone past) unsustainable. 
Think pickling might be the way to go here, one file per device/test as you suggest.

thijstriemstra, 2017-10-31:
Decided json makes more sense. Exporting them now using:

```python
import io
import json
import os.path

from baseline_data import demo_ssd1351

print(demo_ssd1351)

fpath = os.path.join('reference', 'data', 'demo_ssd1351.json')

with io.open(fpath, 'w', encoding='utf-8') as f:
    f.write(json.dumps(demo_ssd1351, ensure_ascii=False))
```

PR #179 (2017-10-31): split baseline data into separate json files
fixes #178
commit 07e30bed (2017-10-31): split baseline data into separate json files (fixes #178) (#179)
closed 2017-10-31 by thijstriemstra