rm-hull/luma.oled #181: MPD Info Scrolling

creisey, 2017-11-09:
Hi there im on a small webradio project with MPD

im struggeling with getting the "Channels Name" info on the SSD1306  SSD.

i've managed to use the python-mpd2 library to display the currentsong['name'],  

but its not updating somehow. probably the variable is not getting resetted.

do you have any suggestions how i could manage that?

> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

RPi0

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 4.9.41+ 

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

thijstriemstra, 2017-11-09:
@creisey does it not make more sense to ask the python-mpd2 project for help with this?

rm-hull, 2017-11-09:
Can you provide a code sample of what you're trying to achieve?

closed 2017-11-20 by thijstriemstra