rm-hull/luma.oled #187: very poor refresh rate with FT232H board, on small img translation (41ko)

elpimous, 2017-12-30:
**Board** : Jetson TX2
**SPI/GPIO controller** : FT232H (adafruit board)
**OS** : Ubuntu 16.06 64 bits
**Screens** : 2 SSD1351 RGB Oled screens (2 in parallel/ same wires(same pins))

Hello, I have a problem : 

As you can see, I had to do some minor changes, to make your libs work with the adafruit board : 
[(https://github.com/rm-hull/luma.oled/issues/185)]

Screens light on with eyes png imgs (41ko), Good.
However, refresh rate is very poor !! (I did a simple picture translation from left to right, (in a simple loop, without pause), and I evaluate FPS to 6 max !!)
```
eye = Image.open('/home/nvidia/Pictures/oeil_transp.png').convert("RGBA")
#upper_eyelid = Image.open('/home/nvidia/Pictures/upper.png').convert("RGBA")
#lower_eyelid = Image.open('/home/nvidia/Pictures/lower.png').convert("RGBA")

moves_number = 8 # number of imgs for each move
delay = 1 # timing before next direction
eyemoveX = -40
destX = 40
moveX = destX-eyemoveX
move_per_loop = (moveX/moves_number)

for i in range(10): # 10 times before closing operation !
  j = 0
  for j in range(moves_number): # 8
    eyemoveX += move_per_loop
    eye_move = np.float32([[1, 0, eyemoveX], [0, 1, move_per_loop]]) # ex 1.0
    lenoeil = eye.transform((128,128), Image.AFFINE, (1, 0, eyemoveX, 0, 1, 0))
    device.display(lenoeil.convert(device.mode))
    j+=1
  eyemoveX = -40
  i+=1
  sleep(delay)
```

I tested your perfloop utility : 
```
perfloop.py -f conf/ssd1351.conf --framebuffer=diff_to_previous

Testing display rendering performance
Press Ctrl-C to abort test
...
2017-12-30 01:35:17,838 - SPI write with command 11.
2017-12-30 01:36:56,306 - SPI write with command 11.= 5024.93 FPS
```

It seems that I have a very good refresh rate, here : why a too poor moving picture FPS ??
I added value to transfer_rate :  transfer_size=14000000 (no changes)

Would you have an idea ??

rm-hull, 2017-12-30:
Firstly, for perfloop, don't use the `diff_to_previous` framebuffer - you need to choose `full_frame` to get a realistic measure of worst-case FPS. The diff-to-previous does exactly what it says - it chooses the bit of the frame that was different to last time, and just renders that - and because the perfloop doesn't change, it doesn't render anything.

In your program, try commenting out the line `device.display(lenoeil.convert(device.mode))` and see what your perceived frame-rate is to isolate other things in your program.

However, I can imagine bridging SPI over USB with the adafruit ft232h python library could be painfully slow: is there a  specific reason why you didn't use the native SPI and GPIO pins on the jetson board?

elpimous, 2017-12-30:
Hi, Richard.
- I have nearly same results with full_frame option : 
```
nvidia@tegra-ubuntu:~$ sudo python /home/nvidia/Downloads/luma.examples/examples/perfloop.py -f conf/ssd1351.conf --framebuffer=full_frame
[sudo] password for nvidia: 
2017-12-30 10:03:56,112 - Disabling FTDI driver.
2017-12-30 10:03:56,112 - Detected Linux
2017-12-30 10:03:56,149 - Called ftdi_usb_open and got response 0.
2017-12-30 10:03:56,149 - Called ftdi_usb_reset and got response 0.
2017-12-30 10:03:56,149 - Called ftdi_read_data_set_chunksize and got response 0.
2017-12-30 10:03:56,150 - Called ftdi_write_data_set_chunksize and got response 0.
2017-12-30 10:03:56,150 - Called ftdi_usb_purge_buffers and got response 0.
2017-12-30 10:03:56,150 - Called ftdi_set_bitmode and got response 0.
2017-12-30 10:03:56,151 - Called ftdi_set_bitmode and got response 0.
2017-12-30 10:03:56,173 - Setting clockspeed with divisor value 29
2017-12-30 10:03:56,176 - Setting clockspeed with divisor value 29
2017-12-30 10:03:56,183 - SPI write with command 11.
2017-12-30 10:05:11,240 - SPI write with command 11.= 4851.25 FPS
```
refresh is very good, isn't it ?

- Ok, I'll investigate the way you propose.

- The reason why I don't use my board, directly : i don't know how to access board GPIO pins with python !! 
(a newbie small problem !) Working on solution here too !

Thanks Richard

elpimous, 2017-12-30:
Well, investigations done : 
On a 50 fps loop (moves_number=50,  in the up python ex),
Commenting the lign : 
#device.display(lenoeil.convert(device.mode))

The loop is immediate. No latency !

Sure that the problem comes from the img sending to oled !
With some more tests , the refresh rate of the simple 40k travelling pict is 1.6/s
I  don t anderstand !! The FT232H is 30Mhz capable
I just need nearly 600ko/s (~15 fps)

Perhaps i did an error/missing part in serial.py ?!

Well....

thijstriemstra, 2018-02-07:
@elpimous any updates?

closed 2023-06-11 by thijstriemstra