rm-hull/luma.oled #189: Weird video artifacts SSD1331 + NanoPi Duo

sgjava, 2017-12-30:
I'm seeing the same issue on the NanoPi Duo as I saw on the CHIP with the SSD1331. This works fine on my ODROID C1+. The I2C SSD1306 works fine on the Duo and I'm going to try the SPI version of the SSD1306 to isolate if it's SPI. The weird thing is everything displays fine until I sleep for 10 seconds.

A video of the iseue:

https://youtu.be/axc_4QPLNPk

rm-hull, 2017-12-30:
When does it start filling in, at the start or end of the 10 second sleep?

Might be worth putting a capacitor across the power rails to smooth things out a bit, close to the GND and VCC pins on the display?

sgjava, 2017-12-30:
Only during the sleep phase after drawing stopped (almost like a flow control issue). The example apps run at full speed, but they also give off weird artifacts. I've always been able to run the SSD1331 off the Pi, ODROID C1, etc. without a cap. Only the CHIP and Duo have behaved like this so far. In any event I'll have a GPIO kit https://www.amazon.com/gp/product/B00HU0Z4C2 that has a dedicated power supply for the rails and comes with 10nF and 100nF ceramic capacitors. Would one of those caps work? I'll let you know the results of the dedicated power test.

CRImier, 2017-12-31:
I also had a similar problem with my SSD1331 (on a Pi Zero, through Chinese prototyping wires), last time I checked. It wouldn't fill the whole screen IIRC, but would do it in blocks instead of lines - I can make a video, too. Unfortunately, I didn't think about adding capacitors - will try it next time I'll be at my workplace!

I can also send one of these - it's likely the same model as in the video, though, and chances are you already have one.

sgjava, 2017-12-31:
Once my breadboard and power supply arrive I will try to power off the PSU rail and not the SBC. SSD1331 works fine on Pi 3, ODROID, etc. directly wired to SBC without any other components.

sgjava, 2018-01-01:
OK, I tested SSD1331 on dedicated power rail using 3V3 PSU like https://www.amazon.com/JBtek-Breadboard-Supply-Arduino-Solderless/dp/B010UJFVTU. Same exact issue. I'm thinking it's the SPI or some kind of flow control issue. I'll look at adding a cap next.

rm-hull, 2018-01-01:
@sgjava: 
* can you post the program listing that is being run in the video?
* if you see weird artifacts in the examples then it is likely that something else is writing to the SPI device at the same time
* is there an alternative SPI port or device you can connect to, eg. if there are both /dev/spi1.0 and /dev/spi1.1 and you have both CE0 and CE1 pins available?
* the only other thing I can suggest is to run some extended [spidev-test](https://github.com/rm-hull/spidev-test) with a loopback wire over MOSI <-> MISO and check that the received data is the same as transmitted data

sgjava, 2018-01-01:
It's the same little test code I use for all my SBCs. The interesting thing is the SPI version of SSD1306 works as expected with the Duo. I've only seen this with the SSD1331 on the Duo and CHIP (it works fine on ODROID C1 and Pi 3 powered off the SBC). This is leading me to believe it's less to do with the SPI although I would not rule that out. Technically with the Duo SPI 0.0 is dedicated to flash and only 1.0 and 1.1 is available. I tried to configure 1.1, but it causes the Duo to act wried (the heartbeat LED stops flashing). I've used spidev to do a MO <-> MI loop back test and it works as expected.

I have 100nF and 10nF capacitors, but I've not seen anyone wire one of these in and I'm a little leery to be the first one :) I've never had to add caps or resistors to these little displays as they have built in voltage generator. You can see where I have the SSD1331 and ST7735 working with the ODROID C1+ here https://forum.odroid.com/viewtopic.php?f=116&t=26674

```
import time
import nanopi.duo
from OPi import GPIO
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1331

GPIO.setmode(nanopi.duo.BOARD)
GPIO.setwarnings(False)
serial = spi(port=1, device=0, gpio=GPIO, gpio_DC=5, gpio_RST=3)
device = ssd1331(serial)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white")
    draw.text((3, 3), "Hello", fill="white")

time.sleep(10)
```

sgjava, 2018-01-01:
Here's the images of SSD1331 and then dropping in SSD1306 to the breadboard with the same configuration. What I'm going to do is see if I can find another SSD1331 library just to rule out any Luma.OLED issues. Then I can look at the power/SPI angles deeper.

[image]
[image]

sgjava, 2018-01-01:
OK, I tried https://gist.github.com/graef685/f1175b9021c5c71734d1566380c3156b and hacked it to work with OPi.GPIO. I get weird stuff flashing, but no circles, etc. One thing I noticed is that the LED on the PSU is dimming slightly when data is sent. The PSU should have maximum output of 700 mA, thus much more than the max 25 mA the SSD1331 uses.

sgjava, 2018-01-01:
OK, doing more research here. https://github.com/sumotoy/SSD_13XX says "RST Connect to +3v3 using a 4k7->10k resistor". Does this mean connect RST to 3V3 rail with 10K Ohm resistor in between? Says the same thing here https://github.com/jrial/ESP8266BitstampTicker. Never had to do that with other SBCs or SSD1306 SPI.

sgjava, 2018-01-01:
OK, tried the 3V3 -> 10K resistor -> RST and the display still freaks out :(

rm-hull, 2018-01-02:
The Raspberry Pi family has built-in pull-up resistors on it's GPIOs so for SPI and I2C applications it is not necessary; for some other SoC devices (e.g. ESP8266) don't, so I would generally add pull-ups. You should research (or ask on the forums) whether the nano-pi has them. 

It may be that if you look at the back of the OLED breakout boards you have there may be some surface mounted 10K resistors and if you can trace the tracks, it may be that these are pull-ups. 

If neither the SBC or the OLED breakout board has pullup resistors, then it is likely to float in an undefined state, which _might_ explain the behaviour you are seeing.

If you have an RPi, NanoPi, SSD1306 and SSD1331, try the all combinations are see which are stable and which aren't: if the RPi works consistently with both OLEDs and the SSD1306 works consistently with both SBCs, this implies that they have pullups, and that the NanoPi and SSD1331 perhaps dont...

Hence, maybe try adding pullups on the SPI chip enable, clock and data lines?

sgjava, 2018-01-02:
On Duo and CHIP:
- SSD1306 SPI/I2C work fine
- SSD1331 freaks out

On Pi, ODROID C1:
- SSD1306 SPI/I2C work fine
- SSD1331 works fine

I only tried 3V3 -> 10K resistor -> RST on Duo, but not SCL and SDA. Will try that. I assume 10K Ohm on each?

rm-hull, 2018-01-02:
yes try a 10K pull-up on each - do all at once

CRImier, 2018-01-06:
@sgjava If you have time before I do, try adding the capacitors - it really might be a power problem (which is even more likely given that, as you say, the PSU LED is slightly dimming). Your kit has those red 10nF and 100nF capacitors - just add one of each (two total), one lead connected to VCC and one to GND (since those capacitors don't seem to be polarized, it doesn't matter which lead goes where).

The imporant thing is that the capacitors need to be as close to the display pins as possible! Plugging them into same rows of the breadboard should work, plugging them into the board you're powering the display from is not what you want.

sgjava, 2018-01-06:
I'm going to try the 10K pull up resistors first. I might get to it today. I have 100nF and 10nF capacitors. Which one should I try if the resistor experiment fails?

CRImier, 2018-01-06:
As I said, put one of each, in parallel (if you only can put one, use 100nF).

sgjava, 2018-01-06:
Yea, sorry, I reread and saw that.

sgjava, 2018-01-07:
Resistors didn't work. Mat try cap next. I'll also try on my ODROID C1 again just to make sure the display is not damaged.

thijstriemstra, 2018-01-07:
@sgjava having a 2nd identical device to test with in this case sounds like a good idea.

sgjava, 2018-01-07:
Haha yea, this particular SSD1331 does work on the ODROID C1, I just want to test it again and make sure I didn't fry anything before I try the capacitor.

CRImier, 2018-01-07:
I've just tested the display again and I can no longer reproduce the problem we're having - with the same hardware that I used before, no capacitors added, no other changes.

sgjava, 2018-01-07:
I've only had this issue with 2 SBCs, The Nano Pi Duo and the CHIP.

rm-hull, 2018-01-07:
Maybe the other thing to try is reduce the SPI bus speed. The examples have a command line argument of `--spi-bus-speed=xxxx`. I think the default is 8000000 (8MHz). Try 1000000 or 500000 to see if that is more stable.

sgjava, 2018-01-07:
Yea, I've tried all that before on the Duo. Almost anything you can think
of config wise. I may order a different OLED that supports SSD1331 commands
since the SSD1306 SPI version works fine.

CRImier, 2018-01-07:
Hmm. Alright, so, I definitely have some issues with one of the displays - not the other that I have, which looks identical. Looks like a power issue, tried to solve it with capacitors, to no avail. Will describe in more detail - a little bit later.

sgjava, 2018-01-07:
Just to be clear I tried another SSD1331 library and had the same issue, so this isn't a Luma problem from what I can tell. Please send a link of where you got the working SSD1331 from. For now I can use it on some of my other SBCs without issue. BTW it looks like it already has the pullup resistors on it, thus adding more cause it not to respond at all.

thijstriemstra, 2018-02-04:
> Just to be clear I tried another SSD1331 library and had the same issue, so this isn't a Luma problem from what I can tell

Closing. Please open a ticket if any issues pop up.

closed 2018-02-04 by thijstriemstra