rm-hull/luma.oled #194: Display issues with SSD1351 128x128 OLED display on Raspberry Pi 3

craver2000, 2018-01-20:
I am having some problems with using the library with my 128x128 SSD1351 display(16-bit Color 1.5" -https://www.adafruit.com/product/1431) on my Rapsberry Pi 3.

For instance, I get distorted or clipped display with demo.py initially. I then tried to change the SPI bus speed settings from 500000 to 16000000 which seemed to help with demo.py by enabling a full frame display but it did not significantly improve the display when I ran other example
scripts. Examples scripts like runner.py fail to work after a spilt second of flicker on the display. For 3d_box.py and crawl.py, I get speckled pixels instead of a clean display during the animation. Any ideas how I can resolve this? I have attached some screenshots. I tested these by running the following command on terminal (and changing SPI speeds each time)
[image]
:
`pi@raspberrypi:~/luma.examples $ python examples/demo.py --display ssd1351 --width 128 --height 128 --interface spi --spi-bus-speed 8000000`
[image]

[image]

[image]

[image]

rm-hull, 2018-01-22:
Looking at the pictures, it almost certainly _looks like_ something else is emitting data on the same SPI channel. 

The fact that you see the warning

```
/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py:164: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  self._gpio.setup(pin, self._gpio.OUT)
```

indicates that there is another program sharing at least one GPIO pin.

Things I would look at to try and isolate the problem:
* Stop all non-standard scripts that you have running
* Change the CS (Chip select) to CE1 (P01-26) and then run the demo with `--spi-device=1`
* Burn a new raspian image on a spare SD card, boot up a clean system & install the libs, and retest

craver2000, 2018-01-22:
Thanks for the suggestions. I change the CS to CE1 first but that did not help. Then, I loaded an early raspbian backup image into the SD card and reinstalled the library, and the demo scripts ran properly this time! While I still have that same runtime warning of potential pin clash, I no longer have the flickering, clipping or artifactual display any more. I tested a couple of scripts and they are running very well. Upon investigating further, I realized that the conflict is caused by an installation of a library for a phat board quite a long time ago, which is no longer used now. Would it be possible for me to resolder the DC connection of the OLED to another pin, say BCM 0 (instead of BCM24), and then I run the demo with `--gpio-data-command 0` to avoid the pin conflict?

craver2000, 2018-01-22:
Ok, just tried using BCM20 for DC and it works!

closed 2018-01-22 by craver2000

rm-hull, 2018-01-22:
Glad you got it sorted 👍