rm-hull/luma.oled #195: Details on python usage

craver2000, 2018-01-23:
I relied on the info here under python usage (https://luma-oled.readthedocs.io/en/latest/python-usage.html#examples) to use the OLED device in my python script. So far in my code, I have used the following snippet of code to initialise display my SS1351 OLED using my raspberry:
```
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1351
serial = spi(device=0, port=0)
device = ssd1351(serial)
device.width=128
device.height=128
```
However, the display appears distorted and I think that is because the changes to the GPIO DC pin to pin 20 is not in the script.
I wanted to ask how can I include this parameter in my script such that changes of my GPIO data command pin to BCM20 of my raspberry pin is taken into account?

I have tried to include `gpio-data-command 20 ` but that does not seem to work.

CRImier, 2018-01-23:
Here's an example of what works for my setup, where both RST and DC are non-default:
```
serial = spi(port=0, device=0, gpio_DC=6, gpio_RST=5)
```

craver2000, 2018-01-23:
Thanks CRImier. Can I also check with you if your other parameters such as the driver, display width and height format are the same as what I used above?

rm-hull, 2018-01-23:
> device.width=128
> device.height=128

You dont need to set that - ssd1351 only supports 128x128 by default

CRImier, 2018-01-23:
It seems like you've deleted a message or something, it appears in my email but not on this page. Anyway, all the other parameters look good to me (and I'm not even using SSD1351, mostly 1106, 1306 and 1331s).

thijstriemstra, 2018-02-04:
@CRImier any news?

thijstriemstra, 2018-02-07:
No feedback, closing. Let us know if it's still not working.

closed 2018-02-07 by thijstriemstra