rm-hull/luma.oled #196: luma.oled working on TinkerBoard + SSD106 0.96"

effebal, 2018-01-25:
This awesome library works on TinkerBoard and luma.OLED.
This is the configuration:

- TinkerOS 2.04
- ASUS GPIO API for python libraries (includes a full GPIO RPi library)
- i2c SSD 1306 0.96" OLED diplay

Here is a short video showing the oled in action.

https://youtu.be/NiKvo_2GfNw

thijstriemstra, 2018-01-25:
Thanks for sharing! That 'tinkerboard' looks **a lot** like a Raspberry Pi ;)

rm-hull, 2018-01-27:
Nice one @effebal !

Was wondering whether we should collate a load of external links showing off what folks have been using the libs for?

thijstriemstra, 2018-01-27:
The wiki is probably a good place for that?

effebal, 2018-01-28:
If I could express my humble opinion, the issue reporting on github is not so good for this kind of news; anyway at the moment it is a good place where sharing.
@thijstriemstra: a dedicated section in the wiki could be the best solution.

thijstriemstra, 2018-02-04:
@effebal can add a link/section to the documentation to points to a wiki page (that we'll create then)?

closed 2023-06-11 by thijstriemstra