rm-hull/luma.oled #205: Problem with SSD1351 . Does not display.

ayhlai, 2018-04-15:
Tried a few examples and cannot see anything on the display. No errors.

I have no problem with i2c displays.

**Checked the SPI port has ok**

ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Apr 12 15:40 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr 12 15:40 /dev/spidev0.1

**The program runs without error, but I see nothing on the display :**

:~/luma.examples$ python examples/perfloop.py -f conf/ssd1351.conf --framebuffer=full_frame
Testing display rendering performance
Press Ctrl-C to abort test

Version: luma.oled 2.3.2 (luma.core 1.7.2)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------
#### iter =   2945: render time = 43.67 ms, frame rate = 22.87 FPS

Hardware: Tried both Pi 3 Model B and Pi 3 Model B+. Same result
OS: Raspbian Release date:2018-03-13 Kernel version:4.9

Connections:
OLED Pin | OLED Name | RPi Pin | RPi Function | Remarks
-- | -- | -- | -- | --
1 | GND | P01-20 | GND | Ground
2 | VCC | P01-17 | 3V3 | +3.3V Power
3 | SCL | P01-23 | GPIO 11 (SCLK) | Serial Clock
4 | SDA | P01-19 | GPIO 10 (MOSI) | Serial Data
5 | RES | P01-18 | GPIO 24 | Reset
6 | DC | P01-22 | GPIO 25 | Data/Command
7 | CS | P01-24 | GPIO 8 (CE0) | Chip Select

Here are some photos. I have 2 displays of the same ssd1351 ic. Both yield the same results (nothing shown on display).

Thanks very much in advance!

[image]
[image]
[image]

rm-hull, 2018-04-22:
I haven't tried a Raspberry 3 or the latest Raspian image; it is entirely possible that some SPI implementation detail changed (unlikely though).

Also since you have two boards, it is unlikely (or very unlucky) that both are dead. Looking at the pic, the soldering doesn't look too good on one of them. Also interesting is that one OLED display is smaller than the other - what size was it? Where did you buy them from?

One thing I found in the past though: using cheap breadboards can cause problems because the metal inserts dont grip the male pins properly (see this [youtube video](https://www.youtube.com/watch?v=_kYdCZwUQMo) about 14mins in). 

I always tend to use the female-female jumper cables like these https://www.ebay.co.uk/itm/10cm-Female-to-Female-Jumper-Wire-for-Breadboard-Prototyping-/262403793547

nanomesher, 2018-05-13:
Finally had time to test this out.  I have took your advice and removed the need for breadboard, directly jumping with cable.

It went one step further, running below:

$ sudo modprobe fbtft_device name=freetronicsoled128
$ dmesg | grep fbtft_device
[   63.900658] fbtft_device: module is from the staging directory, the quality is unknown, you have been warned.
[   63.903323] fbtft_device: GPIOS used by 'freetronicsoled128':
[   63.903337] fbtft_device: 'reset' = GPIO24
[   63.903345] fbtft_device: 'dc' = GPIO25
$ con2fbmap 1 1

then do a "sudo reboot" , I managed to see something on screen, I guess that confirmed wiring is ok.

[image]

However, when I run any luma.examples, I still get a blank display without any error.
eg.
pi@raspberrypi:~/luma.examples $ python examples/maze.py -f conf/ssd1351.conf --framebuffer=full_frame
Version: luma.oled 2.4.0 (luma.core 1.7.2)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------

pi@raspberrypi:~/luma.examples $ python examples/welcome.py -f conf/ssd1351.conf --framebuffer=full_frame
Version: luma.oled 2.4.0 (luma.core 1.7.2)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------

Result: Nothing on display

rm-hull, 2018-05-14:
Ok, what is interesting is that when you have it working with the kernel driver, you have pin 24 connected to reset, and pin 25 connected to the data/command line.

```
[   63.903337] fbtft_device: 'reset' = GPIO24
[   63.903345] fbtft_device: 'dc' = GPIO25
```

However, if you follow the instructions at https://luma-oled.readthedocs.io/en/latest/hardware.html#spi you will see that the _default_ pins for these two lines are the other way around.

Therefore I would suggest either:
* You connect the pins as suggested in the luma.oled documentation
* You maintain your existing pin connections and supply `--gpio-data-command=25` and `--gpio-reset=24` to override the defaults when running the demos.

Either should work.

nanomesher, 2018-05-14:
It's working!! Thanks so much for this brilliant library.

python examples/animated_gif.py -f conf/ssd1351.conf --framebuffer=full_frame --gpio-data-command=25 --gpio-reset=24 --framebuffer=full_frame

[image]

closed 2018-05-14 by rm-hull

chris75vie, 2020-01-06:
For me it's not working unfortunately. 

I'm using following module:
[image]

It works fine with the sample on https://www.waveshare.com/wiki/1.5inch_RGB_OLED_Module

I figured that I connected GPIO-24 for DC and GPIO-25 for RST. Moreover the device is connected to CE0 which is supposedly spi-device 0. I'm using RaspberryPi 4 B and 
`Linux raspberrypi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`

With the luma example the display stays black.

`python examples/demo.py -f conf/ssd1351.conf --framebuffer full_frame --gpio-data-command 24 --gpio-reset 25 --spi-port 0 --spi-device 0` 

Any ideas why luma example does not work although waveshare example does?