rm-hull/luma.oled #206: ssd1327 command exceed 32 bytes

up1021, 2018-05-28:
Traceback (most recent call last):
  File "demo.py", line 99, in <module>
    main()
  File "demo.py", line 54, in main
    device = get_device()
  File "/home/debian/luma.examples-master/examples/demo_opts.py", line 63, in get_device
    device = cmdline.create_device(args)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/cmdline.py", line 179, in create_device
    device = Device(Serial(), **vars(args))
  File "/usr/local/lib/python3.5/dist-packages/luma/oled/device.py", line 705, in __init__
    0xB6, 0x0F,         # 2nd Pre-charge period: 15 clks
  File "/usr/local/lib/python3.5/dist-packages/luma/core/device.py", line 39, in command
    self._serial_interface.command(*cmd)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/interface/serial.py", line 80, in command
    assert(len(cmd) <= 32)
AssertionError

line 591 in file luma.oled.device.py, command exceed 32 bytes.

rm-hull, 2018-05-28:
Ohh, interesting ... and interesting that it works with the tests (it turns out we use a mock for the serial, so this explains why it works). I must admit I accepted PR #202 blind without the ability to test it, as I haven't got an SSD1327 device. 

I will add a fix for this (probably split into a few commands) - would you be prepared to test it please @up1021 ?

closed 2018-05-28 by rm-hull

rm-hull, 2018-05-28:
Released version 2.4.1 with commit 1f93439

commit 1f934399 (2018-05-28): Split SSD1327 init into several commands (#207)

fixes #206