rm-hull/luma.oled #209: Clear screen on draw.text()

freja57, 2018-06-05:
Hi,

After some hesitation and bothering Richard on his private e-mail about it, i am now convinced this is also a place for 'Questions' of 'Help wanted'. My issue cencerning Lumo.Oled is as follows:

Hardware: Raspberry Pi B, rev.2 (for testing)
OS: Linux pitest 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
Python: 2.7.9

I am playing with python on a Raspberry Pi (B rev.2) with A SH1106-controlled Oled display. The Luma package works fine, but examples I encountered seem to suggest that when you want to draw a text, you first clear the area to be used by drawing a rectangle of the appropriate size.
My problem is that each time I draw a text (draw.text) the draw-command first fully clears the display, which erases all other text on the display, unfortunatelly.
Is this normal behaviour as far as you know? Is there a way around it or an argument I have to pass to draw.text? 

#### Expected behaviour
Text appears in previously cleared limited area

#### Actual behaviour
The entire screen is cleared when drawing text

Hope someone recognises this and knows what to do.
Thanks in advance!

Fred Jansen.

CRImier, 2018-06-05:
Hi! Do you have a code sample?

CRImier, 2018-06-05:
> examples I encountered seem to suggest that when you want to draw a text, you first clear the area to be used by drawing a rectangle of the appropriate size.

This is not accurate, you only need to do it when you're not creating a new canvas (`with canvas(device) as draw`).

freja57, 2018-06-05:
Hi. Thanks for your swift reply. your mentioning of creating a canvas made me realize i completely misunderstood the workings of the displaying (new to python and new to luma).
I had a routine to display tekst that looked like this:

def display(r,d,msg,f):
  with canvas(disp) as draw:
    # clear selected area
    size = draw.textsize(msg, font=f)
    draw.rectangle((r,d,r+size[0],d+size[1]),outline=0,fill=0)
    # display new text
    draw.text((r, d), msg, font=f, fill=255)

So each time it is called a new canvas is created which obviously creates an empty canvas. After some more looking for and at examples I found I have to create an image and a drawer, like:

image = Image.new('1', (128, 64), 0)
draw  = ImageDraw.Draw(image)

Which I suppose creates a 128x64 imagespace of black pixels, like a canvas. I hope 
But after putting this in and the removal of the 'with canvas' in my routine all I get is a white screen so far. So it seems I am still missing something.

CRImier, 2018-06-05:
You only get a white screen because the `with canvas...` construct also displays everything once it's done (I was confused by it when I first encountered it, but it became clear after reading [the source](https://github.com/rm-hull/luma.core/blob/master/luma/core/render.py), specifically, the `__close__` method). Try calling `disp.display(image)`.

CRImier, 2018-06-05:
Now, about your routine - it doesn't seem like you'd need to draw a differently-colored rectangle, unless color of your text is the same as color of your background - since, as you mentioned, a new canvas is created each time.

freja57, 2018-06-06:
Added disp.display(image) as you advised. That seems to do the trick! Many thanks!

CRImier, 2018-06-07:
Don't forget to close the issue if the problem is resolved =)