rm-hull/luma.oled #211: Show image and text on the same canvas

dummys, 2018-07-04:
Hi,

I'm trying to show an image at the left of the screen and to show updated stat at the right.
didn't found how to do it using a ppm image.
Can you help me ?
thanks

rm-hull, 2018-07-05:
You can use Pillow's [ImageDraw.bitmap](http://pillow.readthedocs.io/en/5.1.x/reference/ImageDraw.html#PIL.ImageDraw.PIL.ImageDraw.ImageDraw.bitmap) method. 

* Also, have a look at: https://github.com/rm-hull/luma.examples/blob/master/examples/crawl.py
* You can also display font awesome icons - see https://github.com/rm-hull/luma.examples/blob/master/examples/font_awesome.py
 * If the image is unchanging, you could use that as a background: https://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.render.canvas

dummys, 2018-07-06:
Wow, I read the doc but I didnt discover the background option. Thanks for
help.

closed 2018-07-25 by thijstriemstra