rm-hull/luma.oled #212: character flashed in short time

wwhai, 2018-07-14:
when i run this demo:

> from luma.core.interface.serial import i2c, spi
from luma.core.render import canvas
from luma.oled.device import ssd1306, ssd1325, ssd1331, sh1106
serial = i2c(port=1, address=0x3C)
device = sh1106(serial)
with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="black")
    draw.text((30, 40), "Hello World", fill="white")

the OLED display haven't keep longtime,however in a flash,the character disappear.

rm-hull, 2018-07-14:
This behavior is as expected; once the program ends, it will clean up any resources and shutdown the device. So you have two choices:
* add a sleep in a loop to wait indefinitely 
* override the cleanup routine to do nothing, see https://github.com/rm-hull/luma.oled/search?q=cleanup&type=Issues

thijstriemstra, 2018-07-31:
any news @wwhai?

rm-hull, 2018-09-07:
closing - no response

closed 2018-09-07 by rm-hull