rm-hull/luma.oled #216: micropython support

thijstriemstra, 2018-07-31:
I've been using micropython a lot lately and I like the idea of an minimal ESP32 running Python 3.4.

What are the chances of supporting micropython in luma.oled, I guess/hope it's mainly a mapping issue of GPIO ports etc?

Gadgetoid, 2018-07-31:
Although my experience is largely centred on the micro:bit, it gives me some idea of just how complicated writing libraries for a cut down platform can get.

The key problems with MicroPython are:

* It's not Python, no matter how much it looks like it, it's a fundamentally different language that may omit features on the whim of the maintainer because;
* There's no singular MicroPython anymore, everyone's got their own flavour with slightly different device abstraction libraries.
* There is a low recursion limit/call depth based on the system stack size which, at least on the micro:bit, makes some levels of abstraction impossible

And, yes, Pillow wont help things!

A ground-up, MicroPython-specific version of luma with things like support for FRAM display buffers, and targeting one or two versions of MicroPython, and one or two devices with the requisite system resources could work, but I wouldn't be hopeful for direct support.

That said, my experience *has* been with the micro:bit which is MicroPython at its very "micro."

A really good example of point 1 is "slice attribute read access" on the micro:bit:

https://github.com/bbcmicrobit/micropython/blob/da7ce603ff18763f42aacc0318d1118c1a6a65d7/inc/py/mpconfig.h#L737-L741

They're disabled. This means "slice.start" and "slice.stop" don't work for apparently no sensible reason so I had to resort to.

```python
def _s(key):
    return (int(x.replace("slice(","")) for x in str(key).split(",")[:2])

start, stop = _s(slice_object)
```

Yes, you can grab the string representation of a slice object and get the start/stop from that but you can't access them as properties... :cry:

CRImier, 2018-07-31:
What @Gadgetoid said. Even if you can bring Pillow to MicroPython, you're going to encounter a lot of restrictions. I recently had to make a small 8266-powered device - a 16x2 display library, a RC522 library and lots of code in `main.py`. Well, I ended up having to set up a MP toolchain on my laptop and compile both libraries into `.mpy` (or `.mpc` ? ) files, so that they wouldn't throw the device into an `OutOfMemory` error while importing both modules - since MP compiles each cleartext (`.py`) module while importing it, it crashes if there's not enough memory to do so. I also remember sorting out some inheritance fuckery last time I brought my pure Python modules to MP - IIRC attributes of parent class wouldn't be accessible for some reason when introspecting in MP console. That said, it's still an amazing and empowering tool, and I'd pay money to continue using it whenever I feel like it, but you need to be careful when using it. I hope the extended RAM of ESP32 will solve some of these problems.

thijstriemstra, 2018-08-10:
And then there's this: https://github.com/boochow/micropython-raspberrypi/

TCB13, 2022-06-30:
I'm just adding now that there is the new Raspberry Pi Pico W for 6$.

https://www.cnx-software.com/2022/06/30/raspberry-pi-pico-w-a-6-raspberry-pi-pico-board-with-wifi-4/

Raspberry Pi RP2040 dual-core Cortex-M0+ microcontroller @ 133 MHz with 264KB SRAM
Storage – 2MB QSPI flash

It runs MicroPython and C and seems the perfect device to drive displays for cheap applications. I know that there are various limitations with MicroPython but devices like this will become more and more powerful and it would be interesting to see this library running in it.

Meanwhile we've the Raspberry Pi Zero 2 W that brings a proper OS and full support for around 15$.

Thank you all.

closed 2023-06-11 by thijstriemstra