rm-hull/luma.oled #218: Problem about SSD1351 device color format converting

Dreagonmon, 2018-09-11:
Class: luma.oled.device.ssd1351
Funtion: display

Origin: 

    for r, g, b in self.framebuffer.getdata():
        if not(r == g == b == 0):
            # 65K format 1
            buf[i] = r & 0xF8 | g >> 5
            buf[i + 1] = g << 5 & 0xE0 | b >> 3
        i += 2

This code try to convert color RGB(888) to color 65K(565)
Convert: rrrrrrrr,gggggggg,bbbbbbbb -> rrrrr ggg,ggg bbbbb
when deal with the green byte, bit 76543210 should be convert to 765,432
but actually we get 765,210, the green color changed randomly.

The correct code is:

    for r, g, b in self.framebuffer.getdata():
        if not(r == g == b == 0):
            # 65K format 1
            buf[i] = r & 0xF8 | g >> 5
            buf[i + 1] = g << 3 & 0xE0 | b >> 3
        i += 2

green byte just need to left shift by 3, not 5

rm-hull, 2018-09-11:
also note that SSD1331 driver is similarly affected

thijstriemstra, 2018-09-11:
what's the tldr of this bug: basically not seeing correct green tones?

rm-hull, 2018-09-11:
the bug is very _very_ subtle and has virtually negligible effect: basically rather than having 2^6 (=64) shades of green taken from the 6 most significant bits, we have 64 shades of green taken from the 3 most significant bits and the 3 least significant bits. So there might be some odd green-color shifting.

The tests pass locally because the reference test image use solid colors. I thought we might have to re-generate those huge json calls files, but fortunately not...

closed 2018-09-14 by rm-hull

rm-hull, 2018-09-14:
2.5.1 released onto PyPi with this fix in

PR #219 (2018-09-14): Incorrect 16-bit RGB packing
Fixes #218
commit 1e20b025 (2018-09-14): Incorrect 16-bit RGB packing (#219)

- Fixes #218
- Add tests for RGB packing