rm-hull/luma.oled #223: SSD1306 not working via SPI

ski7777, 2018-10-09:
#### Type of Raspberry Pi
tested on both RPi 1 B+ and RPi 3 B+

#### Linux Kernel version
Linux raspberrypi 4.14.70+ #1144 Tue Sep 18 17:20:50 BST 2018 armv6l GNU/Linux

#### Expected behaviour
I expect an image on the display

#### Actual behaviour
Nothing is happening

This is my wiring:
[image]
Please ignore the paper thing. The alignment is a bit wrong...
[image]

Display---RPI
GND---GND
VCC---3V3
DO---SPI0 SCLK
D1---SPI0 MOSI
RES---GPIO BCM 25
DC---GPIO BCM 24
CS---SPI0 CE0

my test code:
```
from luma.core.interface.serial import i2c, spi
from luma.core.render import canvas
from luma.oled.device import ssd1306, ssd1325, ssd1331, sh1106
from RPi import GPIO
GPIO.setmode(GPIO.BCM)
serial = spi(device=0, port=0, gpio=GPIO)
device = ssd1306(serial)
with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="black")
    draw.text((30, 40), "Hello World", fill="white")

input()
GPIO.cleanup()
```

My SSD1306 I²C displays are working with similar code. Why is this not working?

rm-hull, 2018-10-09:
You should let the luma code manage the GPIO to get a working baseline. I highly suspect that it is down to setting the GPIO mode

thijstriemstra, 2021-01-13:
did this ever get resolved..

khwajay, 2021-04-30:
Looks like I have the exact same display and wiring. (I have the same code minus managing GPIO)

Sometimes it flashes white for a millisecond, but most of the time it's a black screen.

DJ-Dingo, 2021-07-30:
If you are using SPI, you should use the gpio_RST and gpio_DC funktions to make it work
Something like this:           serial = spi(device=0, port=0, gpio_DC=16, gpio_RST=18)

where GPIO 16 and 18 on your PI are connected to your display DC and RST or RES

closed 2023-06-11 by thijstriemstra