rm-hull/luma.oled #224: Frame/sec

borsosgy, 2018-10-29:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
RPi3 with Raspbian

#### Linux Kernel version
Linux moode 4.14.72-v7+ #1146 SMP Wed Sep 26 16:58:28 BST 2018 armv7l GNU/Linux

#### Expected behaviour
I want to use SSD1322 OLED from EA with luma.oled.
In OLED datasheet has 135  Frame / sec in Regisrty.  

#### Actual behaviour
The display can not decoded the commands. --> nothing on the screen

How can I set 135 Frame / sec Rate?

Thanks in advence.

Gyula Borsos
gyula.borsos@gmail.com

closed 2018-11-01 by borsosgy

rm-hull, 2018-11-02:
While the device can be driven at 135 FPS, using python on an SBC means that it is almost certainly impossible to get that frame rate with the luma library. There are some benchmarks here:  https://github.com/rm-hull/luma.oled/wiki/Usage-&-Benchmarking ... These should give you a rough idea of likely performance

borsosgy, 2018-11-14:
Hi Richard,

Thanks for your support!

Gyula

Richard Hull <notifications@github.com> ezt írta (időpont: 2018. nov. 2.,
P, 19:29):

> While the device can be driven at 135 FPS, using python on an SBC means
> that it is almost certainly impossible to get that frame rate with the luma
> library. There are some benchmarks here: [...]