rm-hull/luma.oled #225: Images on SSD1306 are displaced right and down

Shirkamdev, 2018-11-04:
I'm testing a SSD1306 display on a Raspberry Pi 3. This board uses SPI. With every example I execute, the drawings appear padded right and down as the middle of the screen. If the demo is animated, drawings start moving down and right, and the part that cannot be drawn due to being out of screen, appears up in the screen.

Here is an example of what's happening:
[image]

I'm launching this example with `python demo.py --interface spi`
I have a running driver, and I've checked all values to see if example defaults are the same, every one of them is correct. I don't know what is happening, as the older driver behaves correctly.

rm-hull, 2018-11-04:
When you said: _"as the older driver behaves correctly"_, what do you mean by that?

There are two possibilities AFAICS:
* Are you certain it is an SSD1306 rather than a SH1106 ?
* Alternatively, it looks as though the memory is remapped from a different offset ... Could you try running the demo with the `--h-offset H_OFFSET` switch, eg. `python demo.py --interface spi --h-offset -32` ... maybe try a few different values to see if you can get it to align properly

Shirkamdev, 2018-11-08:
I'm sure is not a SH1106, if I change the display to that, it does weirder things with the image.
Also, if I use the offset, it doesn't change the image position at all, it's still displaced from the corner.
When I said that "old driver" thing, I'm saying that the screen is working right now with a (crappy) old driver. I'm not sure what can I do to make it work.
Thanks

thijstriemstra, 2018-11-09:
>  I'm saying that the screen is working right now with a (crappy) old driver.

what driver specifically?

Shirkamdev, 2018-11-12:
This one: [Adafruit_Python_SSD1306](https://github.com/adafruit/Adafruit_Python_SSD1306)

thijstriemstra, 2019-08-13:
>  I'm saying that the screen is working right now with a (crappy) old driver. I'm not sure what can I do to make it work.

if it works, it works..?

closed 2019-08-13 by thijstriemstra