rm-hull/luma.oled #227: Why is device a boolean?

mindfuucker, 2018-11-05:
Hi,

I am implementing a display with an ssd1325 controller. I am trying to implement a python script from scratch using the API. 

My display is running perfectly with the following command:
`python3 examples/clock.py --display ssd1325 --interface spi --spi-device=2 --spi-port=1 --gpio-data-command 26 --gpio-reset 12`

In my python script i have the following (just started):

```
from luma.oled.device import ssd1325
from luma.core.interface.serial import spi

def device_setup():
    interface = spi(port=1, gpio_DC=26, gpio_RST=12, device=2)
    device = ssd1325(interface=interface )
    return device

if __name__ == '__main__':
    device = device_setup()
```

The problem is: the device argument in the interface class is a boolean. At first I thought it was PyCharm doing wierd but [the docs also say its a boolean](https://luma-core.readthedocs.io/en/latest/api-documentation.html#luma.core.interface.serial.spi). 

How can I bypass this?

rm-hull, 2018-11-05:
It should be an integer, but the normal values are only usually 0 or 1, and I wonder if somehow this got transcribed as a Boolean in the docstring.

I will fix the docstring and push out a new version

rm-hull, 2018-11-05:
Looks like bool is the type of `cs_high` - a new version of luma.core (1.8.2) should fix this - you may need to `pip install -U luma.core` to fetch the latest once it is published

closed 2018-11-05 by rm-hull

mindfuucker, 2018-11-05:
@rm-hull Thank you

I already decided on using spi1 cs0 with cs0 on pin 16 using:
`dtoverlay=spi1-1cs,cs0_pin=16`