rm-hull/luma.oled #229: Wrong command on ssd1351

irlandezu23, 2018-11-29:
I tried to use your library to write text to a 96x96 OLED display. The problem was the text color on the display was different than what I had specified in the code, also the text was mirrored. I think the bug is in file "device.py". At line 359 you included a bit mask(0x02) called "order" for BGR option. The bit mask should have been 0x04. The other problem is at line 368 where the bit mask is used. There is a logic or between the bit mask and the argument value (0x74 | order). The value 0x74 should have been 0x72. Please consult the DS to completely understand what I am talking about. Basically, you misplaced two bits in that command.

rm-hull, 2018-11-29:
Yes, looking at the datasheet, I think you are right. Thanks for taking the time to investigate and report it.

I will push out a new release for this shortly.

rm-hull, 2018-11-29:
#192 and #201 both seem relevant here

closed 2018-12-01 by rm-hull

rm-hull, 2018-12-02:
v3.0.0 released with a fix for this issue. See https://pypi.org/project/luma.oled/3.0.0/

PR #230 (2018-12-01): Wrong command to init the RGB / BGR color order on SSD1351
Fixes #229
commit e2307d44 (2018-12-01): Wrong command to init the RGB / BGR color order on SSD1351 (#230)

Fixes #229

* Wrong command to init the RGB / BGR color order
* Fix up tests
* Update changelog