rm-hull/luma.oled #233: Confused about hotspots, snapshots and viewports

RBYoureLate, 2018-12-18:
Hello, I currently have[ this Python 3 code ](https://gist.github.com/RBYoureLate/0622f077da774887a33dd26d6b320aa3)for my SSD1322 screen.  The performance is far from optimal. The screen is mainly stationary/static, with only one textbox panning left through part of the screen. I believe I need to use one of the above ^ but I'm struggling to understand the example programs and the documentation given.

Can you please help me to understand them better. Thank you.

rm-hull, 2018-12-19:
> Side note, when I set mode to = "1" I don't get any output to screen at all. Is there a reason for this?

I'll look into this shortly

RBYoureLate, 2018-12-19:
I've moved around the time pauses such that they're no longer inside of the canvas draw block. The animated text moving is still not as smooth as I'd like, it is normally even laggier than this [seen here](https://photos.app.goo.gl/eJzwCfMQuJNCX21FA).

rm-hull, 2018-12-20:
In your gist, you mentioned using a pi zero .. I think you would get a large boost from just using a model 3b or 3b+ instead. Also, in the early days we did a lot of benchmarking, and as I remember python 2 was perceptibly quicker than python 3

RBYoureLate, 2019-01-04:
I've since switched to using [Python 2](https://gist.github.com/RBYoureLate/0622f077da774887a33dd26d6b320aa3) performance still isn't fantastic, the reason I want to use a pi zero is due to it being so small, I'm trying to make a device similar to [this](https://youtu.be/H69WXXRwhxw?t=11) (hence why size is important.) 

The majority of the time most of the screen remains the same, with only [one of the text boxs panning](https://photos.app.goo.gl/eJzwCfMQuJNCX21FA) , is my above code optimising for this?

I've not used any hotspots, snapshots and/or viewports which I thought might be able to help me optimize my code, but I'm confused on their usage if so.

rm-hull, 2019-01-04:
Hi @RBYoureLate, have a look at the demo that @mlos did here: https://github.com/rm-hull/luma.examples/blob/master/examples/image_composition.py

That is probably (from memory) a better way to implement the text box panning

RBYoureLate, 2019-04-15:
Hello, sorry for the long time since I last updated this issue. I have since looked at this example and have made some substantial progress, work can be[ seen here](https://gist.github.com/RBYoureLate/dab4f4b923da648793e453a33849c995). 

There is one issue I'm still having and not sure how to fix, it's to do with panning of[ text vertically as seen in this GIF here](https://photos.app.goo.gl/AgFiLsp4y4nLXdhQA). I can use offset but I need to be able to crop out half of the vertical height, I've tried using image.crop but that is a literal crop and means the offset doesn't work.

At the moment I have both lines of text stacked on top of each other and both moving upwards with a black rectangle covering up the portion below. Then I layer each row, starting with the top row and then printing the middle and finally the bottom, such that each layer is on top of each other. This works at the start as [seen here](https://photos.app.goo.gl/911GT2tGkXVEjaD7A) when I have all three moving, because they are all being updated. 

But I have an issue when I only want to update one, as the layering becomes misplaced and I end up with the row below being black. 

What would be the best way of achieving this? Ideally I'd like to be able to have the text animation contained with a 256 by 16 canvas area.  Thank you for your help, I appreciated I've now deviated far off my original question.

RBYoureLate, 2019-07-07:
I eventually managed to spaghetti together some code by redrawing items again to place them back on top of each other. If you're interested in the finally result you can view [here](https://github.com/jfoot/Bus-Departure-Board).

closed 2019-07-07 by RBYoureLate

thijstriemstra, 2019-07-07:
looks cool!! and runs smooth.