rm-hull/luma.oled #239: ssd1327 on raspberry pi

heiko2402, 2019-01-11:
hi,
how is the right installation on the raspberry pi ?

thanks

mindforger, 2019-01-11:
it is linked in the frontpage of the project

https://luma-oled.readthedocs.io/en/latest/install.html

or what do you mean?

yano999, 2019-01-11:
I also have a problem with ssd1327 128*128 display (tested on 2x RPi and on fresh new instalation - where sh1106 display works fine).
When running demo.py on ssd1327 everything works except "Testing screen updates..." - display gets distorted. On emulator pygame - works great.

thijstriemstra, 2019-01-11:
> display gets distorted

can you post a picture or video?

yano999, 2019-01-12:
Here is video.
Display 128x64 sh1106 i2c and 128x128 ssd1327 spi.
Main difference is at the end and at the beginning as you can see rectangle is much darker on ssd1327.

[20190112_171520.zip](https://github.com/rm-hull/luma.oled/files/2752246/20190112_171520.zip)

rm-hull, 2019-01-13:
@yano999 for some reason the mp4 inside the zip wont play on OS X.

thijstriemstra, 2019-01-13:
@yano999 can you post it on youtube?

CRImier, 2019-01-13:
Here you go: https://www.youtube.com/watch?v=ZRone71Or90

rm-hull, 2019-01-15:
@yano999 
* the SSD1327 is a 4-bit greyscale display, and because the rectangle in the demo is blue, so this renders darker than the red, green and yellow components. If you dont want greyscale rendering, run the demo with `--mode=1`
* The corruption at the end of the video is odd ... how are you running the two demos, both on one RPi? Or are you running them on separate RPi's?
* What spec of RPi are you using? It could be that the SPI bus speed is set to high for your setup

yano999, 2019-01-16:
--mode=1 helped with rectangle. Thank you.

The two demos were running on one rpi in the same time.
Now again just for a test I installed fresh new raspbian with nothing else except described in https://luma-oled.readthedocs.io/en/latest/install.html - result is the same.
I have two (identical) displays on ssd1327, they can be reconfigured with jumper to use i2c - and when I connect it to i2c result is the same...

yano999, 2019-01-16:
I have read luma examples doc's and tried an option:
 --framebuffer full_frame
and it solved the problem in demo.py.
The animation is slower than on sh1106 without this option but it works.
But why ssd1327 needs this option and sh1106 doesn't ?

joske500, 2019-01-24:
I have the exact same model as in the YouTube video; the --framebuffer full_frame did the trick, thanks @yano999!

rm-hull, 2019-03-30:
Fixed in #244. I will release a new pypi package shortly

closed 2019-03-30 by rm-hull

CosmoFox, 2019-03-30:
@rm-hull yo! sorry about question - will you release it today?

rm-hull, 2019-03-30:
Yes!

avvy65, 2019-12-19:
When I run the examples/clock.py script it reports my 1.5" SSD1327 oled display as a  ssd1306 128 x64, even tho' I specified from luma.oled.device import ssd1327. There is nothing on the display

thijstriemstra, 2019-12-19:
> When I run the examples/clock.py script it

How do you run the script and what is the output?

avvy65, 2019-12-19:
pi@raspberrypi:~/luma.examples/examples $ python clock.py
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64

thijstriemstra, 2019-12-19:
Please follow the instructions: https://github.com/rm-hull/luma.examples#installation-instructions

rm-hull, 2019-12-19:
specifically, try: `python clock.py -f conf/ssd1327.conf`

thijstriemstra, 2019-12-19:
If you still have issues, please open a new ticket so we don't spam other users on this thread.