rm-hull/luma.oled #241: Content in portrat mode in RPi oversized & streched

jerzyks, 2019-01-15:
Hello, the display is not able to work properly in portrait mode and the content becomes stretched and oversized. Is there any solution for this problem?

Equipment - Raspberry Pi zero
Interface - SPI
Driver - SH1106
Font - default

Basic script:
```
# Configure the serial port
serial = spi (device = 0, port = 0)
device = sh1106 (serial, rotate = 3)
```

Best regards Jerzy

[image]
[image]

JackJagt, 2019-03-13:
I noticed the same problem on Raspberry py 3
[image]
[image]
Both are stretched

rm-hull, 2019-03-24:
@JackJagt can you post the python program you are using to do the animation please?

thijstriemstra, 2019-08-13:
ping @JackJagt

JackJagt, 2019-08-28:
Hello thijs
I succeeded to solve the problem bij changing  the origen coordidates 

> ping @JackJagt

closed 2019-08-29 by rm-hull