rm-hull/luma.oled #242: ssd1327 and framebuffer='diff_to_previous' not working

kayjayme, 2019-03-24:
My Pi zero and i2c ssd1327 128x128 module shows garbage text when framebuffer='diff_to_previous' is used on the demo.py example. I'd like to use it for faster partial display updates

CosmoFox, 2019-03-24:
Have same config like a @kayjayme

rm-hull, 2019-03-24:
Right, looking back over some of the issues around SSD1327 (see [here](https://github.com/rm-hull/luma.oled/search?q=ssd1327&type=Issues)), it turns out I accepted a PR (#202) from @sharkusk for the SSD1327 implementation. I don't actually have one of these devices to test against unfortunately. If anyone is able to donate or loan me a display that would help resolve this issue.

This issue is a duplicate of #239, cc @yano999, joske500, heiko2402, mindforger

kayjayme, 2019-03-24:
I can give you one - where should I send it

rm-hull, 2019-03-24:
Send me an email to rm_hull@yahoo.co.uk, and I’ll let you know where to post it to.

closed 2019-03-30 by rm-hull

rm-hull, 2019-03-30:
Version 3.1.1 released with a fix the `diff_to_previous` framebuffer, but I get a sense that performance is not great, even with this. Clearly the upper limit of 400KHz on the I2C interface affects things, and setting the `mode=1` to force monochrome will marginally improve things, but I'm not sure how much more can be done to improve the frame rate when there are large changes between frames

CosmoFox, 2019-03-30:
> Version 3.1.1 released with a fix the `diff_to_previous` framebuffer, but I get a sense that performance is not great, even with this. Clearly the upper limit of 400KHz on the I2C interface affects [...]

For now it`s much more faster. Fine for my project. 
Mode=1 havent affect fps. 
Tnx for your work and fix,

PR #244 (2019-03-30): SSD1327: Adjust column addressing
Fixes #242
commit 77606b25 (2019-03-30): SSD1327: Adjust column addressing (#244)

Fixed #242 

* Set correct column address
* Micro-optimizations
* Fix typo
* Change back to list
* Amend expected test results based on different command arguments
* Fix flake8 warnings