rm-hull/luma.oled #243: SSD1322 random dots

javierin, 2019-03-25:
#### Type of Raspberry Pi
RPi3

#### Linux Kernel version
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

#### Expected behaviour
I try to run an example to test a display and all I get are apparent random dots all over the screen. The display is a 256x64 monochrome, yellow SSD1322 OLED as reported by the ebay seller:
https://www.ebay.com/itm/3-12-oled-display-module-spi-interface-256x64-ssd1322-for-arduino-stm32-51-uk/122859863076

The command lines I have tried to run the example:
python examples/animated_gif.py --conf conf/ssd1322.conf
python3 examples/animated_gif.py --conf conf/ssd1322.conf
(installed dependencies using both pip and pip3)

I also tried adding some recommended lines to ssd1322.conf, which seem to make no difference:
--display=ssd1322
--interface=spi
--width=256
--height=64
--spi-bus-speed=16000000
--mode=1
--framebuffer=diff_to_previous
--spi-device=0

#### Actual behaviour
This are 2 pictures of the board and the actual output
https://imgur.com/a/TZ6HNEg

rm-hull, 2019-03-26:
* Do  the dots appear before the program starts, or only when the program is run?
* Try a much lower SPI bus speed - 500,000 to start, to see if that works any better
* Do the dots change when a program runs?
* You seem to have a lot of jumper wires plugged in. You should only need max 7 wires, roughly as per https://luma-oled.readthedocs.io/en/latest/hardware.html#spi

javierin, 2019-03-27:
Hi

- The dots appear whenever the example is run and clear whenever I stop it. 
- I tried lowering, but not as much as 500k, will give it a run in a few days and report back
- The dots are static though the program execution. I have found several "patterns" to be consistent across program runs and rpi reboots, so I assume they are not all that random
- I tried wiring only the basic wires, saw the random dots and followed a suggestion to connect all unused wires to ground. In both cases the results were the same.

I will update in a few days about the 500k bus speed. Thanks!

javierin, 2019-03-27:
I got it working. For other's reference, I finally ended up with the following wiring:
2 GND labeled wires
D0 to SCLK
D1 to MOSI
VCC to 5V
CS to SPI0 CS0
D/C to GPIO 23

I left the ssd1322 conf as follows:
--display=ssd1322
--interface=spi
--width=256
--height=64
--spi-bus-speed=500000

Also, it was crucial to indicate the --gpio-data-command to the GPIO I had connected D/C to, in my case this line along with the config and the proper wiring makes the display work nice.

pi@raspberrypi:~/luma.examples $ python3 examples/demo.py -f conf/ssd1322.conf --gpio-data-command 23

Thanks @rm-hull

closed 2019-03-27 by javierin