rm-hull/luma.oled #247: font size how to

riker65, 2019-04-16:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

riker65, 2019-04-16:
Hi
wondering how to change font size here?
Thanks for hints
T

rm-hull, 2019-04-16:
what have you tried so far?

riker65, 2019-04-16:
actually how to add new fonts to your libraries?

sorry , I am a beginner.

Thanks

RBYoureLate, 2019-04-16:
```
FontTemp = ImageFont.truetype("./TestFont.otf",16)
draw.text((0, 0), "Example Text", font=FontTemp)
```
You can try something similar to this, you will obviously need to change the font to a file you have on your system and place this into the appropriate canvas drawing.

riker65, 2019-04-17:
thanks 
works solved

closed 2019-04-17 by riker65