rm-hull/luma.oled #249: Problem with luma/core/interface/serial.py

cartman343, 2019-04-21:
**Setup:**
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/
Raspberry Pi 2 Model B V1.1 (2014)
1.5inch RGB OLED Module (128x128)
connected via SPI

**Test-Code:**
`from time import sleep
import RPi.GPIO as GPIO
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1327
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont

serial = spi(device=0,port=0)
device = ssd1327(serial)

font = ImageFont.truetype("/home/pi/luma.examples/examples/fonts/pixelmix.ttf",12)

with canvas(device) as draw:
    draw.rectangle(device.bounding_box, outline="white", fill="black")
    draw.text((20,40), "Hello World", font=font, fill="white")
    draw.text((20,60), 'Cool Stuff', font=font, fill="white")
    print("in")
sleep(10)
print("Stop")`

**Console Output:**
`Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "copyright", "credits" or "license()" for more information.
>>> 
====================== RESTART: /home/pi/test_oled3.py ======================

Warning (from warnings module):
  File "/home/pi/.local/lib/python3.5/site-packages/luma/core/interface/serial.py", line 164
    self._gpio.setup(pin, self._gpio.OUT)
RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
in
Stop
>>> `

It leans on your example from the link below:
https://luma-oled.readthedocs.io/en/latest/python-usage.html

rm-hull, 2019-04-21:
You haven't said what the problem is.
Are you talking about the channel in use warning?
Is anything displayed on the screen?

cartman343, 2019-04-22:
Hi, 
problem is screen displayed nothing.
Channel warning fixed with `device.cleanup()`.
Tested now with this code:
`from time import sleep
from PIL import Image
from PIL import ImageDraw
from PIL import ImageFont
from luma.core.interface.serial import spi
from luma.core.render import canvas
from luma.oled.device import ssd1327

serial = spi(device=0,port=0)
device = ssd1327(serial)

image = Image.new("RGB", (128, 128), "BLACK")
draw = ImageDraw.Draw(image)

font = ImageFont.truetype('FreeSans.ttf', 15)
draw.text((20,40), "Hello World", font=font, fill="white")
draw.text((20,60), 'Cool Stuff', font=font, fill="white")

device.display(image)
sleep(10)
device.cleanup()`
Sry if it is to easy for you but i still not understand where the problem is.
Pin connection is correct, like in your documentation. 
OLED works fine with the example from waveshare:
https://www.waveshare.com/wiki/File:1.5inch_RGB_OLED_Module_Code.7z

Thx

rm-hull, 2019-04-22:
If you clone the Luna.examples git repo and try and run the demo.py script. Paste the output of that command here and we will at least know what version of the code you are running. Also a photo of your wiring (showing the pins of the OLED and the RPI) will also help

rm-hull, 2019-04-22:
> 1.5inch RGB OLED Module (128x128)
> connected via SPI

What makes you think the RGB device connected over SPI you have is actually an SSD1327?
The only SSD1327 I have seen is a greyscale device over I2C.

Where did you buy the device from ? ...  please provide a link!

Are you sure it isn’t an SSD1351 ?

cartman343, 2019-04-23:
Yes, you got it. You are right, its an ssd1351. That was the problem. 
I've picked the false datasheet of the 1.5inch OLED Module.
Now it works. Thanks you! I'm sorry that I took your time.

closed 2019-04-23 by thijstriemstra