rm-hull/luma.oled #250: Contention with two OLEDs on one SPI controller

TAMHAN, 2019-04-25:
#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
RPI3 B Classic

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

I want to run two different LCDs off the same SPI bus while reusing the DC and RST lines, simply via SPICE0 and SPICE1. The schematic works under Android Things, but not under Raspbian with LumaOLED. 

We need to work with two code revisions: GIRA0 being
serial = spi(device=0, port=0, gpio_DC=16, gpio_RST=18)
device = ssd1306(serial)
serial2 = spi(device=1, port=0, gpio_DC=16, gpio_RST=18)
device2 = ssd1351(serial2)

And Gira1 being
serial2 = spi(device=1, port=0, gpio_DC=16, gpio_RST=18)
device2 = ssd1351(serial2)
serial = spi(device=0, port=0, gpio_DC=16, gpio_RST=18)
device = ssd1306(serial)

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

With Gira0, only the ssd1351 works. With Gira1, only the ssd1306 works. I already scoped the signals, and upon a quick visual check, saw no delta. 
[image]

Oddly, using different DC and RST GPIO pins for the two displays makes the solution work also with the Luma OLED library.

I do have a logic analyzer and can run a trace if you so desire. Or, you can visit my lab in Budapest and play around with the schematic,

TAMHAN, 2019-04-25:
Fotgot: CH1 is D0, CH2 is D1, 3 is RST and 4 was DC

rm-hull, 2019-04-26:
* Can you clarify what GIRA0 and GIRA1 mean? are they just names for your software / hardware?
* What version of luma.oled and luma.core are you using?
* How did you install luma.oled and luma.core (e.g. pypi or git clone)?
* Can you provide a schematic how you have wired up the two devices - specifically what pins you are connecting to on the RPi to which pins in the two displays
* What output do you see if you type the following at the command line: `ls -l /dev/spidev*` ?
* Can you provide a more complete code example that demonstrates the issue you are trying to resolve

TAMHAN, 2019-04-28:
Hello,
sorry for my slow response. This code is owned by a customer, so I can only email it to you - email me at tamhan@tamoggemon.com if you want. I will, however, respond to what I can:

GIRA0 and GIRA1 are two versions of the code.

I think I used pip, not git clone.

I am working on the schematics.

The output is:
pi@raspberrypi:~ $ ls -l /dev/spidev*
crw-rw---- 1 root spi 153, 0 Apr 26 06:10 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Apr 26 06:10 /dev/spidev0.1

larojas, 2020-04-25:
> Oddly, using different DC and RST GPIO pins for the two displays makes the solution work also with the Luma OLED library.

Actually if you just use separate RST pins, it will work (i.e. you can share the DC line). 

In theory it should be possible to operate both screens with only separate CS lines so there's something in either the spi or oled modules that results in only the last created device working. If I find a fix I'll post it here.

retango, 2022-03-12:
Hi! Wondering if this issue was solved.. I'm experiencing the same problem. Two spi devices on same "port" but with different "device" numbers. Both devices connected to all of the same pins, except for CS (Chip Select). Only the device that was created latest turns on. Thanks!

TAMHAN, 2022-03-13:
Hello,

sorry - I did not track this down any further.

Tam

With best regards
Tam HANNA

Enjoy electronics? Join 19k7 other followers by visiting the Crazy Electronics Lab athttps://www.instagram.com/tam.hanna/

retango, 2022-03-13:
Hi, Thanks for the quick reply.. I was able to solve it with my setup (4x ssd1362 connected to the Rpi, sharing all connectors except for CS, of which I have one for each oled). 
It seems the problem is that when instantiating a new "serial" object, it's __init__ function sends a reset signal that turns off the devices that were created before. I solved it by creating the new serial objects after the initial one with the parameter "gpio_RST=None", which avoids this reset signal. My code looks like this:

device0=ssd1362(spi(device=0))
device1=ssd1362(spi(device=1, gpio_RST=None))
device2=ssd1362(spi(device=2, gpio_RST=None))
device3=ssd1362(spi(device=3, gpio_RST=None))

and now I can work with the 4 devices at the same time. I guess this only works if all 4 oleds are connected when the initial device object is created. Hope this helps other people. 

Also a quick note, I'm sure this is not the right place but maybe it's helpful. In the luma documentation it says:
"Because CS is connected to CE0, the display is available on SPI port 0. You can connect it to CE1 to have it available on port 1. If so, pass port=1 in your serial interface create call."
I believe there is a mistake, you should pass "device=1", instead of "port=1", when all connectors but CE are shared. 

Thanks to all for all the help and for creating this amazing package!

closed 2023-06-11 by thijstriemstra