rm-hull/luma.oled #251: ssd1325 spi device fails to correctly update screen

stylerf, 2019-04-30:
#### Type of Raspberry Pi
Pi 3 Model B

#### Linux Kernel version
```Linux raspberry 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux```

#### Expected behaviour
Run examples without issue. Was previously working with same hardware configuration. SD card reformatted with fresh raspbian image, and updating canvas no longer works as expected. Verified hardware setup, and all examples ran with previous software install.

#### Actual behaviour
Running ```demo.py``` from ```luma.examples``` with:
```
$ python3 luma.examples/examples/demo.py -d ssd1325 --interface spi --gpio-reset 22
Version: luma.oled 3.2.0 (luma.core 1.8.3)
Display: ssd1325
Interface: spi
Dimensions: 128 x 64
------------------------------------------------------------
Testing basic canvas graphics...
Testing contrast (dim/bright cycles)...
Testing display ON/OFF...
Testing clear display...
Testing screen updates...
```
First few Tests display as expected, but fails on ```Testing screen updates...```. See gif below.  

[image]

Similar results are observed from other examples that should update. Running ```clock.py``` with:
```
$ python3 luma.examples/examples/clock.py -d ssd1325 --interface spi --gpio-reset 22
Version: luma.oled 3.2.0 (luma.core 1.8.3)
Display: ssd1325
Interface: spi
Dimensions: 128 x 64
------------------------------------------------------------
```
Results in garbled output, shown below.  
  
[image]

rm-hull, 2019-04-30:
A nearly identical issue was reported for SSD1327 devices (see #242) and was recently fixed in #244. I wonder if the same regression has crept into the SSD1325 driver also. The reason this happens is because of the way the column addressing is being set when using the diff to previous frame buffer. 

The issue will “go away” if you add the `—frambuffer full_frame` flag when you run the examples.

In the meantime, I will investigate further and provide a fix for the diff-to-previous framebuffer implementation. It could be something to do with the logic here: https://github.com/rm-hull/luma.core/blob/master/luma/core/framebuffer.py#L52 but this hasn’t changed in a long time.

closed 2019-04-30 by rm-hull

rm-hull, 2019-04-30:
@stylerf  - v3.2.1 released to pypi with a fix for this

stylerf, 2019-04-30:
Thanks for the super quick reply and fix! That did the trick! 

I was able to use the examples with the ```--framebuffer full_frame``` flag. Upgrading to luma.oled v3.2.1 via pip removes the need for the flag. You rock!

PR #252 (2019-04-30): Use correct column addressing for SSD1325
Fixes #251
commit c423e8fa (2019-04-30): Use correct column addressing for SSD1325 (#252)

Fixes #251