rm-hull/luma.oled #253: ioctl(self.fd, I2C_RDWR, ioctl_data) OSError: [Errno 22] Invalid argument

iv-med, 2019-05-17:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
Raspberry Pi Zero W

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
Linux raspberrypi 4.14.98+ #1200 Tue Feb 12 20:11:02 GMT 2019 armv6l GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

Traceback (most recent call last):
  File "LinuxProject4.py", line 55, in <module>
    device = ssd1327(serial)
  File "/usr/local/lib/python3.5/dist-packages/luma/oled/device/__init__.py", line 651, in __init__
    nibble_order=1, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/luma/oled/device/greyscale.py", line 36, in __init__
    self.clear()
  File "/usr/local/lib/python3.5/dist-packages/luma/core/mixin.py", line 46, in clear
    self.display(Image.new(self.mode, self.size))
  File "/usr/local/lib/python3.5/dist-packages/luma/oled/device/greyscale.py", line 118, in display
    self.data(list(buf))
  File "/usr/local/lib/python3.5/dist-packages/luma/core/device.py", line 46, in data
    self._serial_interface.data(data)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/interface/serial.py", line 106, in data
    self._bus.i2c_rdwr(self._i2c_msg_write(self._addr, [self._data_mode] + data))
  File "/home/pi/.local/lib/python3.5/site-packages/smbus2/smbus2.py", line 497, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 22] Invalid argument

rm-hull, 2019-05-18:
Have you followed ALL the pre-requisites here: https://luma-oled.readthedocs.io/en/latest/hardware.html#i2c ?

What is the output of `i2cdetect -y 1` ?

iv-med, 2019-05-18:
> Have you followed ALL the pre-requisites here: https://luma-oled.readthedocs.io/en/latest/hardware.html#i2c ?

yes

> What is the output of `i2cdetect -y 1` ?
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

rm-hull, 2019-05-18:
I'd seen that error before, and from what I remember it gets generated when a device doesn't respond on a particular address. Obviously, as you can see, there is something on 0x3C (this is the usual address for OLED displays).

What is the output of `pip list | grep luma` ?

Also, can you share a bit more of your code, specifically the bit where you initialize the `serial` device?

iv-med, 2019-05-18:
> I'd seen that error before, and from what I remember it gets generated when a device doesn't respond on a particular address. Obviously, as you can see, there is something on 0x3C (this is the usual address for OLED displays).
> 
> What is the output of `pip list | grep luma` ?

pi@raspberrypi: $ pip3 list | grep luma
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
luma (0.9.6)
luma.core (1.9.0)
luma.oled (3.2.1)

> Also, can you share a bit more of your code, specifically the bit where you initialize the `serial` device?

from luma.core.interface.serial import i2c, spi
from luma.core.render import canvas
from luma.oled.device import ssd1327

serial = i2c(port=1, address=0x3C)
device = ssd1327(serial)

rm-hull, 2019-05-19:
When I run an SSD1306 via I2C on the latest versions of luma.core and luma.oled I don't experience any errors, but when I tried the SSD1327 I got **exactly** the same problem as you described.

If I downgrade to luma.core 1.8.3 (`sudo pip install luma.core==1.8.3`) then ithe SSD1327 display works as expected. I would recommend downgrading to the earlier viersion  in the meantime, until we can get to the bottom of the issue.

I think this regression is caused by this commit: https://github.com/rm-hull/luma.core/commit/3502326a48886d2340e713b31de061d1f376ca84 ... @Gadgetoid - are you able to shed any light on this at all?

iv-med, 2019-05-19:
> When I run an SSD1306 via I2C on the latest versions of luma.core and luma.oled I don't experience any errors, but when I tried the SSD1327 I got **exactly** the same problem as you described.
> 
> If I downgrade to luma.core 1.8.3 (`sudo pip install luma.core==1.8.3`) then the SSD1327 display works as expected. I would recommend downgrading to the earlier version in the meantime, until we can get to the bottom of the issue.
> 
> I think this regression is probably caused by this commit: [rm-hull/luma.core@3502326](https://github.com/rm-hull/luma.core/commit/3502326a48886d2340e713b31de061d1f376ca84) ... @Gadgetoid - are you able to shed any light on this at all?

Okay, thank you, after downgrading it works.

rm-hull, 2019-05-20:
[Will keep this open until bug is fixed]

rm-hull, 2019-05-24:
Reading the docs for [`ioctl`](https://docs.python.org/2/library/fcntl.html#fcntl.ioctl) (the method call that actually fails) it seems like there is a 1024 buffer limit, so I'm betting that this is the cause of this error.

closed 2019-05-25 by rm-hull

rm-hull, 2019-05-25:
Luma.core 1.10.1 released with the fix for this issue. Use `sudo pip install luma.core -U` to update to latest.