rm-hull/luma.oled #254: Does it support SSD1326?

iggymayer, 2019-05-19:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

rm-hull, 2019-05-19:
No, SSD1326 is not presently supported, only because we do not have a device to test against. If you (or anyone else) wants to fund or donate/loan a display to add support for this device please get in touch.

thijstriemstra, 2020-11-22:
why close this?