rm-hull/luma.oled #255: luma.oled 3.2.1 identifies ssd1327 as ssd1306 gives black screen

A-n-o-r-a-k, 2019-05-24:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.
Raspberry pi zero W

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

Linux raspberrypi 4.19.42+ #1219

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.
when running tv_snow.py with a ssd1327 I expect the program to identify the oled as a ssd1327 with 128x128 resolution. 

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).
what actually happens is the ssd1327 is identified as a ssd1306 with 128x64 and the tv_snow.py program runs but nothing is displayed on the ssd1327 screen.

I have tested with a ssd1306 attached and it works fine. I have verified the ssd1327 is working by testing on another project. I have tried 7 ssd1327's all are identified as ssd1306

this just started happening after the most recent update.

rm-hull, 2019-05-24:
If you are using the luma.examples, the demo scripts aren't smart enough to figure out which display you are using, so you should generally specify the device, interface and width/height. There is a config flag and some predefined options (see https://github.com/rm-hull/luma.examples/tree/master/conf). 

So you should start the examples with:

```console
$ python examples/tv_snow.py -f conf/xxx.conf
```
Where `xxx` is your display device. I thought I had added one for SSD1327, but looks like I haven't so I'll do that shortly.

The default options on the examples are to select SSD1306 and I2C. If you look at the readme there are many options you can tweak for your device, but I will make it clearer that if no options are set it will default to SSD1306. 

Also, for the short term you should follow the advice in the comments of #253 until we sort out the bug.

rm-hull, 2019-05-24:
@A-n-o-r-a-k when you run the tv-snow example, could you provide the command you use and console output please when you run the examples.

A-n-o-r-a-k, 2019-06-06:
@rm-hull tried it again with the -f conf file and with several ssd1327 screens and the tv_show.py program does not throw an error, I get the prompt back and nothing on the screen. I have checked all of the wiring and confirmed the screen works on another project. I am going to try to back rev the package as described above.

rm-hull, 2019-06-06:
Uploading a photo of the wiring going from the screen to the raspberry pi would help. 
The latest luma.core package fixes the bug mentioned in #253, so you should make sure it is up-to-date with `pip install luma.core -U`

thijstriemstra, 2019-07-07:
any feedback @A-n-o-r-a-k?

thijstriemstra, 2020-02-14:
no feedback, closing.

closed 2020-02-14 by thijstriemstra