rm-hull/luma.oled #256: SSD1327 IOError: [Errno 6] No such device or address

sonelu, 2019-05-26:
#### Type of Raspberry Pi
>  NanoPi Core2

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.
Linux PD2 4.14.111 #2 SMP Tue Apr 30 16:04:14 CST 2019 aarch64 aarch64 aarch64 GNU/Linux

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.
I've been using an SSD1306 without any problems and I have changed to an SSD1327. The device is recognised by the kernel:
```
pi@PD2:~$ i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- 1e -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --     
```
The display is at address 0x3c as the previous one. 0x1e and 0x6a are a gyro/accelerometer and 0x70 a GPIO expander.

I've updated the `luma.core` to 1.10.1:
```
pi@PD2:~$ pip list | grep luma
luma.core (1.10.1)
luma.oled (3.2.1)
```
When trying to initialise the screen I get this:

```
pi@PD2:~$ python
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from luma.core.interface.serial import i2c
>>> from luma.oled.device import ssd1327
>>> serial = i2c(port=0, address=0x3c)
>>> display = ssd1327(serial)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/luma/oled/device/__init__.py", line 651, in __init__
    nibble_order=1, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/luma/oled/device/greyscale.py", line 34, in __init__
    self._init_sequence()
  File "/usr/local/lib/python2.7/dist-packages/luma/oled/device/__init__.py", line 663, in _init_sequence
    0xA8, 0x7F)         # set multiplex ratio: 127
  File "/usr/local/lib/python2.7/dist-packages/luma/core/device.py", line 39, in command
    self._serial_interface.command(*cmd)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/interface/serial.py", line 85, in command
    list(cmd))
  File "/home/pi/.local/lib/python2.7/site-packages/smbus2/smbus2.py", line 482, in write_i2c_block_data
    ioctl(self.fd, I2C_SMBUS, msg)
IOError: [Errno 6] No such device or address
```
Thanks.

sonelu, 2019-05-26:
Ignore the issue. I will close it.
Seems to be a hardware problem on the bus with the new devices that I have installed.

Thanks.

closed 2019-05-26 by sonelu

rm-hull, 2019-05-26:
I was going to ask whether the NanoPi core 2 has built-in pull-up resistors on the I2C data and clock lines. If not, you may want to add a 10K pull-up resistor on each line.

sonelu, 2019-05-26:
The NanoPi doesn't but the display is connected through an add-on board that has the pull ups. As a matter of fact both the screen and the other GPIO extender boards have their own pull ups. So I removed the pair from the screen but that did not solve the problem. In the end II have separated the GPIO and the screen and I have tested only with the screen. That works fine.

So I will continue to see where the problem is. I have another GPIO board that I can test separately independent on the screen and then try again all of them on the same bus. There are a few other hacks that I did on the boards - I might have screwed up something - although I've triple checked and nothing seems to be a problem.

Thing is, the `luma.oled` is working fine.

Thanks for the work, and for sharing it by the way.