rm-hull/luma.oled #258: Support for SH1122 (256 x 64)?

LongLane, 2019-07-16:
Hi, great work. I was wondering if you are planning to support SH1122 OLED displays? Here is an [example](https://uk.farnell.com/midas/mcot256064dy-wm/oled-graphic-display-tab-256x64pixels/dp/3154954?st=MCOT256064D)

Happy to sponsor one for you.

rm-hull, 2019-07-16:
@LongLane Always open to support new devices. Happy to accept screens sent direct to me or donations for devices gratefully received. This one https://www.aliexpress.com/item/32846176247.html may be better as it is on a breakout board and is configured for SPI (the farnell one looks to be I2C, so would be quite a bit slower)

Ping me at rm_hull@yahoo.co.uk and I'll let you know the mailing address or alternatively paypal to that email and I'll order direct from aliexpress

thijstriemstra, 2020-02-14:
@LongLane still looking for sponsors, can you give an update?

LongLane, 2020-02-14:
Hi, good point, this one had slipped. I had sent a sample over to Richard. @rm-hull, do you have an update? Many thanks

LL

poulsen93, 2022-01-18:
Hey @rm-hull, did you ever get display to play with?
I currently looking at this one: [AliExpress](https://m.aliexpress.com/item/4000868235850.html?browser_id=96e1948b2af845288aa5925999dc1bb2&aff_trace_key=d113c6d41dc842618328fa0bf4dac87c-1634154555781-04255-_py7F1P3&aff_platform=msite&m_page_id=17e69e0d33a9b075f6ab8616f856fc3def7124d35f&gclid=) 
And would happily donate one to you if your still interested.

sukovec, 2022-11-21:
Hello. I see no progress here, but I managed to have one of these displays working with this library. You can find it [here](https://github.com/sukovec/rpi-rackplay/blob/276e222d9b5e82d9c748fbf6e46bf42abfc7d130/display/sh1122.py) - it's working for me (4 wire SPI), but the code is very messy.

It does not use the possibility to update changed parts of canvas as this display probably does not support same commands as other oled drivers do (or I haven't found it). Feel free to use the code. 

Biggest thanks should go here: https://forum.micropython.org/viewtopic.php?f=14&t=9769

kvandt, 2023-10-20:
I'd love to use my SH1122 display with luma.oled. Anyone managed to integrate above code of @sukovec in this repo? I'd love to test.

sinnfeinn, 2023-12-10:
Hi, i have no idea if anyone can help here but Im trying to get this display to work on raspberry pi and am somewhat on the way but need some help:

**sh1122.py:**
```
import spidev
import RPi.GPIO as GPIO
import time

# Constants for the SH1122 display
SET_COL_ADR_LSB = 0x00
SET_COL_ADR_MSB = 0x10
SET_DISP_START_LINE = 0x40
SET_CONTRAST = 0x81
SET_SEG_REMAP = 0xA0
SET_ENTIRE_ON = 0xA4
SET_NORM_INV = 0xA6
SET_MUX_RATIO = 0xA8
SET_CTRL_DCDC = 0xAD
SET_DISP = 0xAE
SET_ROW_ADR = 0xB0
SET_COM_OUT_DIR = 0xC0
SET_DISP_OFFSET = 0xD3
SET_DISP_CLK_DIV = 0xD5
SET_PRECHARGE = 0xD9
SET_VCOM_DESEL = 0xDB
SET_VSEG_LEVEL = 0xDC
SET_DISCHARGE_LEVEL = 0x30

class SH1122:
    def __init__(self, width, height, spi, dc, res, cs):
        self.width = width
        self.height = height
        self.spi = spi
        self.dc = dc
        self.res = res
        self.cs = cs
        self.buffer = bytearray(self.width * self.height // 8)
        self.init_display()

    def init_display(self):
        # Hardware reset
        GPIO.output(self.res, GPIO.HIGH)
        time.sleep(0.010)  # 1ms delay
        GPIO.output(self.res, GPIO.LOW)
        time.sleep(0.010)  # 10ms delay
        GPIO.output(self.res, GPIO.HIGH)

        # Initialization sequence for the display
        commands = [
            SET_DISP | 0x00,  # Display off
            SET_COL_ADR_LSB,
            SET_COL_ADR_MSB,
            SET_DISP_START_LINE | 0x00,
            SET_SEG_REMAP,
            SET_MUX_RATIO, self.height - 1,
            SET_COM_OUT_DIR,
            SET_DISP_OFFSET, 0x00,
            SET_CONTRAST, 0x80,  # You might need to adjust this
            SET_ENTIRE_ON,
            SET_NORM_INV,
            SET_DISP | 0x01,  # Display on
        ]
        for cmd in commands:
            self.write_cmd(cmd)

    def write_cmd(self, cmd):
        GPIO.output(self.cs, GPIO.LOW)
        GPIO.output(self.dc, GPIO.LOW)
        self.spi.writebytes([cmd])
        GPIO.output(self.cs, GPIO.HIGH)

    def write_data(self, data):
        GPIO.output(self.cs, GPIO.LOW)
        GPIO.output(self.dc, GPIO.HIGH)
        self.spi.writebytes(data)
        GPIO.output(self.cs, GPIO.HIGH)

    def display(self, image):
        if image.mode != '1':
            raise ValueError("Image must be in mode 1.")

        # Convert image to 1-bit data
        width, height = image.size
        pixels = list(image.getdata())
        buffer = bytearray(width * height // 8)

        for y in range(height):
            for x in range(width):
                if pixels[x + y * width]:
                    page = y // 8
                    buffer[x + width * page] |= (0x80 >> (y % 8))

        self.buffer = buffer
        self.show()

    def show(self):
        self.write_cmd(SET_COL_ADR_LSB)
        self.write_cmd(SET_COL_ADR_MSB)
        for page in range(self.height // 8):
            self.write_cmd(SET_ROW_ADR + page)
            start = page * self.width
            end = start + self.width
            self.write_data(self.buffer[start:end])

    def clear(self):
        self.buffer = bytearray(self.width * self.height // 8)
        self.show()

```
**test.py:**
```
import spidev
import RPi.GPIO as GPIO
from PIL import Image, ImageDraw, ImageFont
from sh1122 import SH1122
import time

WIDTH = 256  # OLED display width
HEIGHT = 64  # OLED display height

# SPI and GPIO setup
spi = spidev.SpiDev()
spi.open(0, 0)  # SPI bus 0, device 0 (can adjust as needed)
spi.max_speed_hz = 100000  # Adjust as needed

GPIO.setmode(GPIO.BCM)
dc = 26    # Data/Command
res = 6   # Reset
cs = 18    # Chip Select

GPIO.setup(dc, GPIO.OUT)
GPIO.setup(res, GPIO.OUT)
GPIO.setup(cs, GPIO.OUT)

# Initialize the display
oled = SH1122(WIDTH, HEIGHT, spi, dc, res, cs)

# Create an image for the Raspberry Pi logo
logo = Image.new('1', (32, 40))
# Here, you need to draw the Raspberry Pi logo onto 'logo'
# The provided byte array can be used to draw the logo

# Create a drawing object
draw = ImageDraw.Draw(logo)
# Draw the logo (replace with actual logo drawing code)

# Create a blank image for drawing
# This will be used for text and blitting the logo
image = Image.new('1', (WIDTH, HEIGHT))
draw = ImageDraw.Draw(image)

# Clear the display
oled.clear()

# Draw the logo on the display
image.paste(logo, (112, 0))

# Add some text
font = ImageFont.load_default()
draw.text((5, 5), "Raspberry Pi", font=font, fill=255)
draw.text((5, 15), "Pico", font=font, fill=255)

if HEIGHT > 32:
    draw.text((5, 25), "This display", font=font, fill=255)
    draw.text((5, 35), "has more than", font=font, fill=255)
    draw.text((5, 45), " 32 lines", font=font, fill=255)

if WIDTH > 128:
    draw.text((149, 25), "This display", font=font, fill=255)
    draw.text((149, 35), "has more than", font=font, fill=255)
    draw.text((149, 45), " 128 pixels", font=font, fill=255)

# Transfer the image to the OLED buffer and display it
oled.display(image)

# Add a loop to keep the display on
try:
    while True:
        time.sleep(1)
except KeyboardInterrupt:
    # Clean up
    spi.close()
    GPIO.cleanup()
```

it displays text but kind of garbled. see image attached.
[image]

Maybe someone can help!

IamGPIO, 2024-09-15:
@rm-hull I have a SH1122 I am desperate to get working with the Luma.OLED Library. Would you be interested in looking at this? Happy to send you a screen and pay for your time. I too am in the UK. 

Let me know if you can help. Thanks

Ezlanding1, 2025-01-07:
Can confirm the SH1122 driver (sh1122.py) works perfectly with everything I've tested it with using the [MELIFE SH1122 256x64 OLED SPI](https://www.amazon.com/MELIFE-Display-Module-SH1122-256x64/dp/B0CZDLC59J). Thank you @sukovec !!

rickyrickuk, 2025-06-05:
> > > Can confirm the SH1122 driver (sh1122.py) works perfectly with everything I've tested it with using the [MELIFE SH1122 256x64 OLED SPI](https://www.amazon.com/MELIFE-Display-Module-SH1122-256x64/dp/B0CZDLC59J). Thank you [@sukovec](https://github.com/sukovec) !!
> > 
> > 
> > Hi, can you advise where the sh1122.py file is please? Trying to get a display working too. Thanks
> 
> The sh1122 driver made by sukovec can be found [here](https://github.com/sukovec/rpi-rackplay/blob/c4127a50f0f49081e691911940111b9e34e5b496/display/sh1122.py), and it can be used like any other luma oled device (https://luma-oled.readthedocs.io/en/latest/python-usage.html)

That's great, thanks so much.
Downloaded and all working perfectly :-)