rm-hull/luma.oled #259: OLED remains blank when script is run automatically

doceave, 2019-08-10:
Hi there

Raspberry Pi Zero W with fresh install of Buster, just recently updated and upgraded :)
Luma library also fresh download :)

#### Hardware setup
> Raspberry Pi Zero W
> Linux raspberrypi 4.19.58+ #1245 Fri Jul 12 17:20:08 BST 2019 armv6l GNU/Linux

#### Expected behavior
> I have a machine I have just finished prototyping which has an OLED screen driven by your SH1106 library
> This OLED performs admirably when python script is run from terminal manually
> Python script is executable
> I would like my script to execute automatically at boot 
> I would like the screen to function as it would during prototyping  

#### Actual behavior
> Sadly on boot the script runs and all aspects of the machine are functional except the screen which remains blank
> If the script is run manually the OLED behaves as it should.
> I have attempted to run automatically by means of Crontab and rc.local
  - Both exhibit the same problem.

I am totally puzzled. I don't even have any error logs to get clues from as no errors are raised (and I have no error handlers in my script which could be mopping them up)

Simply summarized: Script set to run automatically results in blank OLED.

Please do share any leads you may have.

Dylan.

thijstriemstra, 2020-02-14:
can you share the script/cronjob etc @doceave?

thijstriemstra, 2020-04-07:
no feedback, closing.

closed 2020-04-07 by thijstriemstra