rm-hull/luma.oled #260: displacement of images in the SSD1306 display in the raspberry pi 3 using SPI

pgarciaaisoy, 2019-08-14:
I'm using a SSD1306 128x64 OLED display in a raspberry pi 3 model through SPI. Although most of the examples run correctly all the images seems to move from left to right. Here a link showing the issue: https://www.icloud.com/photos/#07_CJ9J5Xswey9HbgvnvuuDvw
(Sorry for the comments in Spanish)

previously we used the Adafruit SSD1306 drivers and that didn't happen.

Do you have any ideas of what could be happening? am I doing something wrong?

Thanks

Pablo

mindforger, 2019-08-14:
can you upload that foto to github instead of linking it from icloud?

pgarciaaisoy, 2019-08-16:
Hello,
Thanks for contacting me. It is a large video (aprox. 100mb). Is it ok for you if I give you a link from another service? Eg Dropbox?

Best regards,

Pablo García Martínez
CTO, Aisoy Robotics

m: +34 644436054
w: aisoy.com e: pgarcia@aisoy.com

thijstriemstra, 2019-08-16:
if it's not a picture, but a video, upload it to youtube or vimeo.

pgarciaaisoy, 2019-08-17:
https://vimeo.com/354394553

Password: luma

Thanks

Best regards,

Pablo García Martínez
CTO, Aisoy Robotics

m: +34 644436054
w: aisoy.com e: pgarcia@aisoy.com

closed 2023-06-11 by thijstriemstra