rm-hull/luma.oled #261: Adafruit 1.3" display (ADA938) 'snow' displayed

baisey, 2019-09-17:
Hello
Im trying to use the above OLED to show audio information from Moode Audio. 
I tested it using the adafruit libraries. All examples worked correctly.
I'd like to use pydPiper, the output i get looks like this - 
https://github.com/rm-hull/luma.oled/issues/70

I believe pydPiper uses the luma.oled drivers.
I installed luma.oled and get the same output using this command - 
sudo python weather.py --i2c-address 0x3D --i2c-port 1 --display ssd1306
I've also tried --display sh1106 with the same result.
All luma.oled example files show the same 'snow'

Raspberry PI 3
Connected via i2c
i2c address (derived from i2cdetect) - 0x3D
Moode audio distribution

Any advice would be much appreciated.
thank you
Baisey

thijstriemstra, 2019-12-19:
@baisey i'm not sure what piedpiper is. Can you test with our supported OS instead: Raspbian.

baisey, 2019-12-19:
Hello thijstriemstra
Thanks for the reply.
Yes, i will test using Raspbian. Hopefully i'll get some time over Christmas.
To add some more information - 
This is pydPiper - https://github.com/dhrone/pydPiper
This is the guide i used  - http://moodeaudio.org/forum/showthread.php?tid=461
Interestingly, on boot with pydPiper running as a service i'd get the snow display. I noticed if i ran an adafruit demo the display would reset and show moode song information. I thought it might be something to do with initializing the display so I edited one of the adafruit examples.
https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/examples/shapes.py
deleted everything after line 81. Saved a new script and run this on boot and the display works.
This may or may not be useful.
Thanks
Baisey

thijstriemstra, 2020-02-14:
> Yes, i will test using Raspbian. Hopefully i'll get some time over Christmas.

any news @baisey?

baisey, 2020-02-25:
Hello
Sorry it took a while.
I installed Raspbian (release date - 2020-02-13).
Installed luma.oled as per instructions on readthedocs and all examples worked perfectly. So i'd say there are no problems with luma.oled. 
I still think its something to do with resetting the oled but i'll have to keep searching.
thanks
Baisey

closed 2020-02-25 by rm-hull