rm-hull/luma.oled #262: "Python usage" didn't work, examples work fine

ed-french, 2019-09-19:
Platform: Raspberry Pi 4
Display: SSD1306 128x64 monochrome oled 1bpp 
Connection: i2c (Channel 1)
Address: 0x3c

(tests repeated with second (different type) display and behaved identically)

Followed install instructions precisely (several times).
Followed instructions in python usage.
Correctly identified display as SSD1306
Nothing displays
No error messages reported

Downloaded "Terminal" example and "Clock" example together with the demo_opts.py
Both run correctly.

> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

ed-french, 2019-09-20:
I think I found out my problem. When the program ends immediately following the example code the display is blank, however if I add a sleep statement on the end, the message appears as expected.
Please consider adding a message to the documentation on usage and add the sleep statement to the end of the example.

rm-hull, 2019-09-20:
This behaviour is by design: when the program ends, it runs a cleanup job that puts the OLED display in a low-power state (which effectively turns the screen off) and closes any open devices/channels.

If you don't want this happening either (1) keep the program running, or (2) have a look at issue #140 , and also: rm-hull/luma.core#74

closed 2019-09-20 by rm-hull

ed-french, 2019-09-20:
Sorry I wasn't clear enough, I wasn't suggesting any change to the behaviour, but to the documentation. Not knowing this was the behaviour meant I spent a ton of time trying to find out why nothing (appeared) to happen. I've put in a suggested change to the wording.