rm-hull/luma.oled #265: Pimoroni 1.12” I2C oled

toldcodger, 2019-12-07:
> Firstly thanks for taking the time to file an issue. Chances are, you are not the first (and
> wont be the last) to have this problem: By creating a public issue, others can browse and
> solve their issues - possibly in a self-service style - just by reading the discourse on your
> ticket, so try and be clear in the describing the problem you have.
>
> Secondly, github issues are not _just_ for raising problems - if you have a question, a
> documentation fix, or suggested improvement, please get in touch.
>
> Lastly, there are a number of related LUMA projects - please check to make sure to create
> the issue in the right GitHub repository.

#### Type of Raspberry Pi
> Not all Pi's are equal at the hardware level - what works on one, might not work on the next.
> This library has been tested on every variant except the RPi3.

#### Linux Kernel version
> Paste in the output from running `uname -a` at the command line on your Pi.

#### Expected behaviour
> Add a few concise notes about what you are expecting to happen.
> Even better, if you paste in a code sample that demonstrates what you want to achieve.

#### Actual behaviour
> Now add some details about what actually happened - if there is an unexpected crash, paste in the
> [traceback info](https://www.google.com/search?q=python+traceback#q=What+is+a+traceback%3F).

toldcodger, 2019-12-07:
Hi,

I’m using a Raspberry Pi 3B+ with a Pimoroni 1.12” I2C oled display.
Most of the examples work properly and I’m really pleased with the results, but the Welcome example does nothing.

There are no error messages, the command below runs, but the screen is blank.  Any assistance gratefully received.

python3 welcome.py --display sh1106 --height 128  --rotate 2 --interface I2C

thijstriemstra, 2019-12-07:
maybe specify the `--width` as well?

toldcodger, 2019-12-07:
Thanks for the quick response.
After I press enter it shows the basic info below:
Version : luma.oled 3.3.0 (luma.core 1.12.0)
Display: sh1106
Interface: I2C
Dimensions: 128 x 128
It tried the width option and the info was exactly the same.
I changed the width to 64 and it said 64 x 128 was not a valid screen size.
So I set the width back to 128 and the height to 64 .
It now shows something in the bottom left quarter of the screen.  I thought the height and width being reversed was something to do with rotate.  So I tried rotate 0,1, 2 and 3.  1 or 3 show nothing.  0 or 2 show something in the bottom left quarter of the screen.
With no height specified it uses a 128x64 display
If I do not specify height or width it also uses a 128 x 64

thijstriemstra, 2019-12-07:
can you add a link to the datasheet of the device? what resolution does it actually support?

toldcodger, 2019-12-07:
https://www.displayfuture.com/Display/datasheet/controller/SH1107.pdf

toldcodger, 2019-12-07:
The instructions on the Pimoroni website say to use the luma library

thijstriemstra, 2019-12-07:
it's an SH1107, not a SH1106.. not sure if it makes a lot of difference. thoughts @rm-hull?

toldcodger, 2019-12-07:
I've just set it up on a RP3A+ it has the same issue with Welcome but after a short while it gives this error:

pi@raspberrypi:~/luma.examples/examples $ python3 welcome.py --display sh1106 --rotate 2 --interface i2c
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: sh1106
Interface: i2c
Dimensions: 128 x 64
------------------------------------------------------------
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.5/dist-packages/luma/core/threadpool.py", line 24, in run
    func(*args, **kargs)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/virtual.py", line 208, in paste_into
    super(snapshot, self).paste_into(image, xy)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/virtual.py", line 172, in paste_into
    self.update(draw)
  File "/usr/local/lib/python3.5/dist-packages/luma/core/virtual.py", line 187, in update
    self._fn(draw, self.width, self.height)
  File "welcome.py", line 166, in render
    size = draw.multiline_textsize(t, font)
  File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 280, in multiline_textsize
    line_width, line_height = self.textsize(line, font)
  File "/usr/lib/python3/dist-packages/PIL/ImageDraw.py", line 273, in textsize
    return font.getsize(text)
  File "/usr/lib/python3/dist-packages/PIL/ImageFont.py", line 140, in getsize
    size, offset = self.font.getsize(text)
OSError: invalid composite glyph

thijstriemstra, 2019-12-07:
run the example from the root of the repository, not the `examples` dir or the font cannot be found.

try

```
python examples/welcome.py --display sh1106 --rotate 2 --interface i2c
```

toldcodger, 2019-12-07:
your example results in a 128 x 64 display with the text showing in the bottom left quarter as before.  adding --height 128 results in a blank screen as before.
I get the error after a short while.  The basic difference is your example looks as though it is running Python 2,  my previous error was using Python 3.

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/luma/core/threadpool.py", line 24, in run
    func(*args, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 208, in paste_into
    super(snapshot, self).paste_into(image, xy)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 172, in paste_into
    self.update(draw)
  File "/usr/local/lib/python2.7/dist-packages/luma/core/virtual.py", line 187, in update
    self._fn(draw, self.width, self.height)
  File "examples/welcome.py", line 166, in render
    size = draw.multiline_textsize(t, font)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageDraw.py", line 280, in multiline_textsize
    line_width, line_height = self.textsize(line, font)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageDraw.py", line 273, in textsize
    return font.getsize(text)
  File "/usr/lib/python2.7/dist-packages/PIL/ImageFont.py", line 140, in getsize
    size, offset = self.font.getsize(text)
IOError: invalid composite glyph

toldcodger, 2019-12-07:
I’m sorry but I’m going to have to go.  Once again life is getting in the way of enjoyment 😁. I’ll keep checking this post and try anything you might come up with.  I think his may just a basic incompatibility between the hardware and the software though.
If the library wasn’t written for this device I think it’s amazing that this is the only real problem.
Thanks for all your assisance

closed 2023-06-11 by thijstriemstra