rm-hull/luma.oled #266: Where do I send a ticket regarding the luma project

avvy65, 2019-12-19:
I sent a ticket to the main Github group, but they didn't build the luma project, so I need to send it to the luma group.
Thanks

thijstriemstra, 2019-12-19:
> main Github group

where is that??

what is the issue, luma.oled related?

avvy65, 2019-12-19:
I have a MakerHawk 1.5" OLED 128 x 128 display and it uses the SSD1327 driver, which I cannot install from AdaFruit as it is not on the Pypi site. Anyway, I edited the luma.examples/examples/clock.py script:
from luma.oled.device import ssd1327

When I then run the script, it returns;
pi@raspberrypi:~/Downloads/luma.examples/examples $ python3 clock.py
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: ssd1306
Interface: i2c
Dimensions: 128 x 64

and nothing is displayed

avvy65, 2019-12-19:
Hi,
       yes it is luma related and I was told by rm-hull to submit a ticket
earlier today

Regards
Rob

thijstriemstra, 2019-12-19:
Please follow the instructions: https://github.com/rm-hull/luma.examples#installation-instructions

and run: `python clock.py -f conf/ssd1327.conf`

avvy65, 2019-12-19:
I have already tried that and this is the output:
pi@raspberrypi:~/Downloads/luma.examples/examples $ python clock.py -f
conf/ssd1327.conf
Traceback (most recent call last):
  File "clock.py", line 70, in <module>
    device = get_device()
  File "/home/pi/Downloads/luma.examples/examples/demo_opts.py", line 56,
in get_device
    config = cmdline.load_config(args.config)
  File
"/usr/local/lib/python2.7/dist-packages/luma.core-1.12.0-py2.7.egg/luma/core/cmdline.py",
line 116, in load_config
    with open(path, 'r') as fp:
IOError: [Errno 2] No such file or directory: 'conf/ssd1327.conf'

thijstriemstra, 2019-12-19:
oh then:

```
cd ~/Downloads/luma.examples/
python clock.py -f conf/ssd1327.conf
```

avvy65, 2019-12-19:
pi@raspberrypi:~/Downloads/luma.examples $ python clock.py -f
conf/ssd1327.conf
python: can't open file 'clock.py': [Errno 2] No such file or directory

I do have a conf folder here:
/Downloads/luma.examples/conf $
and the SSD1327.conf file is there

avvy65, 2019-12-19:
I just checked  / tried the final part of installation :
finally install the luma libraries
pi@raspberrypi:~/Downloads/luma.examples $ sudo -H pip install -e

Usage:
  pip install [options] <requirement specifier> [package-index-options] ...
  pip install [options] -r <requirements file> [package-index-options] ...
  pip install [options] [-e] <vcs project url> ...
  pip install [options] [-e] <local project path> ...
  pip install [options] <archive url/path> ...

-e option requires an argument

thijstriemstra, 2019-12-19:
first:

```
cd ~/Downloads/luma.examples/
sudo -H pip install -e .
```

then:

```
python ~/Downloads/luma.examples/clock.py -f ~/Downloads/luma.examples/conf/ssd1327.conf
```

avvy65, 2019-12-19:
Ok, now that made a difference.:
pi@raspberrypi:~/Downloads/luma.examples $ python
~/Downloads/luma.examples/examples/clock.py -f
~/Downloads/luma.examples/conf/ssd1327.conf
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: ssd1327
Interface: i2c
Dimensions: 128 x 128

but then
clock.py: error: I2C device not found on address: 0x3C

i2cdetect - 1 returns no address at all for the display, so I guess it is
faulty

thijstriemstra, 2019-12-19:
> i2cdetect - 1 returns no address at all for the display, so I guess it is
faulty

try i2cdetect with and without sudo to see if it makes a difference. Also try 0 instead of 1.

avvy65, 2019-12-19:
Thanks but I also tried it on an arduino uno using a i2c scanner sketch and
it didn't din't find an i2c address.
I have sent it back now, and will look for another one to work with my pi
zero.

Regards
Rob

thijstriemstra, 2019-12-19:
okidoki. Feel free to re-open this ticket if you have software issues with that one.

closed 2019-12-19 by thijstriemstra