rm-hull/luma.oled #268: Trying to add support for the ssd1362 (ELW2106AA) screen

xcarcelle, 2019-12-30:
Hi there,

thanks a lot for your great project and support. I have been adding a basic support for the ssd1362 oled screen (Futaba ELW2106AA) but the result is not great yet.

- Hardware :
The screen is connected to a Rpi3 with 3-wire SPI connexion (MOSI, CLK, DC, RST, CS), it is powered with 16V for the OLED power supply and 3V3 for the VCI (Logic Supply).
- Software :
The screen reacts to "luma.example" but only displaying a white line moving slowly from top to bottom.

```
~/dev/luma.examples $ python3 examples/demo.py --conf conf/ssd1362.conf 
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: ssd1362
Interface: spi
Dimensions: 256 x 64
------------------------------------------------------------
Testing basic canvas graphics...
Testing contrast (dim/bright cycles)...   <-- screen starts to react
Testing display ON/OFF...
Testing clear display...
Testing screen updates...
```

The RPi3 is currently running raspiab-buster 
```
uname -a
Linux pi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
```

Below is a small video of the current rendering (I have tried with two different new screens and same results).

[image]

Any hardware/software ideas can be tested on the current set-up to add a working support to the luma.oled project.

Thanks again for this great project.

thijstriemstra, 2020-01-02:
See pull request #267

thijstriemstra, 2020-01-22:
#271 was merged and released (3.4.0), closing.

closed 2020-01-22 by thijstriemstra