rm-hull/luma.oled #270: Waveshare SSD1351 display stays black using luma example

chris75vie, 2020-01-11:
Related to https://github.com/rm-hull/luma.oled/issues/205
Using a different hardware manufacturer it's not working unfortunately. 

I'm using following module:
[image]

It works fine with the sample on https://www.waveshare.com/wiki/1.5inch_RGB_OLED_Module

I figured out that I connected GPIO-24 for **DC** and GPIO-25 for **RST**. Moreover the device is connected to CE0 for **CS** which is supposedly spi-device 0. I'm using RaspberryPi 4 B and 
`Linux raspberrypi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`

With the luma example the display stays black.

```
pi@raspberrypi4:~/oled/luma.examples $ sudo python examples/demo.py -f conf/ssd1351.conf --framebuffer=full_frame --gpio-data-command=24 --gpio-reset=25 --spi-port=0 --spi-device=0
Version: luma.oled 3.3.0 (luma.core 1.12.0)
Display: ssd1351
Interface: spi
Dimensions: 128 x 128
------------------------------------------------------------
Testing basic canvas graphics...
Testing contrast (dim/bright cycles)...
Testing display ON/OFF...
Testing clear display...
Testing screen updates...
```

Here another configuration difference: In the waveshare example the baudrate is 9MHz :
```
#SPI init
SPI = spidev.SpiDev(0, 0)
SPI.max_speed_hz = 9000000
SPI.mode = 0b00
```
whereas in the ssd1351.conf it's 16Mhz. I tried 8MHz without a difference.

Any ideas why luma example does not work although waveshare example does?

rm-hull, 2020-01-11:
Could you take a picture of the cables as connected to RPi please?

chris75vie, 2020-01-12:
[image]

chris75vie, 2020-01-17:
Is there a good way to debug the luma example? In fact the example views no exception or error.

tiberg, 2020-01-20:
I was having the exact same problem with the same display. Thanks for posting this. I zoomed in on your photo and saw it is actually a SSD1351 (the number is printed in reverse on the ribbon connector) so I looked at mine and sure enough, it's also an SSD1351. Not what I bought. So I changed the parameter and it works! yay first time I got this to work.

rm-hull, 2020-01-21:
Only thing I can suggest is to doublecheck the wiring, including the quality of the wires. It is more than likely to do with that. Have you tried a different driver as discussed in #205?

tiberg, 2020-01-22:
I have the same setup. I'm using the same SDD1351 and verified your wiring matches mine. Mine works. I agree. Check that each wire isn't broken. If you don't have a multimeter, then replace each wire one at a time until something happens. Here are the parameters I use:
[image]

sudo python3 demo.py -d ssd1351 -i spi --width 128 --height 128 --bgr

closed 2021-01-13 by thijstriemstra