rm-hull/luma.oled #272: OLED 128x64 Garbled pixels on bottom RH Corner of screen

Dougdv, 2020-02-03:
Hi, I hope you can help me, I am using a VL53l0x  to measure distance and record the value on the oled via a switch statement, the gar
[
bling appears after the value in each switch case is display. See Video and code.

Video:   https://youtu.be/RBn5p17udHQ

Bowles_Measure__WORKING_3_field.zip](https://github.com/rm-hull/luma.oled/files/4145732/Bowles_Measure__WORKING_3_field.zip)
Regards Doug

xcarcelle, 2020-02-03:
Hi Doug,
thanks for your issue. Could you try with a Rpi3 and i2c-connected ssd1306 screen and a equivalent of ```display.setCursor(100, 46)``` code with luma libraries to discriminate the screen and the Adafruit_SSD1306 library ?
Regards,
Xavier.

Dougdv, 2020-02-03:
I am sorry perhaps I should have mentioned I am using Arduino Nano 328P,  I2C connection on both VL53l0x and the OLED display

thijstriemstra, 2020-02-14:
>  I am using Arduino Nano 328P, I2C connection on both VL53l0x and the OLED display

wait.. ping @rm-hull

closed 2023-06-11 by thijstriemstra