rm-hull/luma.oled #273: OSError: [Errno 121] Remote I/O error SH1106 display

Pada69, 2020-02-11:
Hello everybody,

I have a problem with the oled SH1106 module on raspberry pi 3B +.

When I try the examples, it tells me the following message:

`Version: luma.oled 3.4.0 (luma.core 1.12.0)
Display: sh1106
Interface: i2c
Dimensions: 128 x 64
------------------------------------------------------------
Traceback (most recent call last):
  File "font_awesome.py", line 161, in <module>
    main()
  File "font_awesome.py", line 156, in main
    draw.text((left, top), text=code, font=font, fill="white")
  File "/usr/local/lib/python3.7/dist-packages/luma/core/render.py", line 43, in __exit__
    self.device.display(self.image)
  File "/usr/local/lib/python3.7/dist-packages/luma/oled/device/__init__.py", line 125, in display
    self.data(list(buf))
  File "/usr/local/lib/python3.7/dist-packages/luma/core/device.py", line 46, in data
    self._serial_interface.data(data)
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 119, in data
    write(list(data[i:i + block_size]))
  File "/usr/local/lib/python3.7/dist-packages/luma/core/interface/serial.py", line 128, in _write_large_block
    self._bus.i2c_rdwr(self._i2c_msg_write(self._addr, [self._data_mode] + data))
  File "/usr/local/lib/python3.7/dist-packages/smbus2/smbus2.py", line 637, in i2c_rdwr
    ioctl(self.fd, I2C_RDWR, ioctl_data)
OSError: [Errno 121] Remote I/O error`

While with the SSD1306 screen, no display issues.

Has anyone ever had this problem?

Thanks by advance for your return,

xcarcelle, 2020-02-11:
Hi @Pada69 
from the log output it seems that you may an issue with the i2c bus connecting the rpi with the oled screen, could you check with a linux i2c tool such as i2cdetect with such commands
```
i2cdetect -y 0
i2cdetect -y 1
```
and validate that the i2c address detected on the buses are the address of the sh1106 (it should be 0x3C from docs but maybe 0x3D). Can you print the result of the commands ?
Regards,
Xavier.

Pada69, 2020-02-12:
Hi @xcarcelle,

Thanks for your return. After more investigation, we use a shield on the raspberry pi who power supply display with 5V on VCC, SCL and SDA. When we have bought the diplay, it's was indicated 3.3 - 5 V  power range supply. 

But the datasheet indicate Vdd1 1.65 -3.5V and Vdd2 3.0 - 4.2V.

When we plug the display with 5v and start an oled.examples, the beginning of the examples print on the screen but after 0.5 second, the error message appear. And when we add "device.persist = True" the screen keep the last image print on it.

We have solved the problem with power supply the screen with 3.3v on VCC, SCL and SDA.

thijstriemstra, 2020-02-12:
> We have solved the problem with power supply the screen with 3.3v on VCC, SCL and SDA.

Good to hear!

closed 2020-02-12 by thijstriemstra