rm-hull/luma.oled #275: ellipse() doesn't work on sh1106 1.3'' OLED screen

Slaveche90, 2020-02-13:
I'm using 1.3'' OLED screen with sh1106 driver chip. It is 128x64 pixels screen.
I'm trying to run examples/demo.py 
Everything seems to work except ellipse() function. It does nothing on the screen.
`with canvas(device) as draw:`
` draw.ellipse((4, 20, 30, 30), outline=255, fill=255)`
I get no errors, and it does not matter which values of the argumens I use, ellipse isn't diplayed on the screen. 
All other functions for shapes work nice. 

And, when rotating the screen with, rotate = 0, 1, 2 or 3 in the following lines of code:
`device = sh1106(serial, rotate=1, width=128, height=64)`
` with canvas(device) as draw:`
`     draw.rectangle(device.bounding_box, outline=255)`
The image at rotate = 1 or 3 gets moved to one corner, not rotating it properly, image below:
[image]

Slaveche90, 2020-02-14:
I did not read the docs correctly.. ellipse works great.
ellipse((X1, Y1, X2, Y2), outline=255, fill=255)
X1 and Y1 is the top left point of the rectangle around the ellipse, and X2 and Y2 is the bottom right point of the rectangle around the ellipse

thijstriemstra, 2020-08-24:
any idea @rm-hull?

Joshyakadamien, 2023-02-24:
Bump? this is still as it is described...