rm-hull/luma.oled #277: Installation of luma.oled

bighegi, 2020-03-08:
Hi there,
I try to install luma.oled.
I worked successfully through the prerequisites for an sh1106 i2c oled.
However, trying to install luma.oled does not work:

`sudo -H pip3 install --upgrade luma.oled
Collecting luma.oled
  Using cached https://files.pythonhosted.org/packages/b0/fa/ef1065ad3644608d6127ce05da43b2b6ab1857b5b26983439f696bab6f4a/luma.oled-3.4.0-py2.py3-none-any.whl
Collecting luma.core>=1.12.0 (from luma.oled)
  Using cached https://files.pythonhosted.org/packages/a7/36/f10ac3202221c479bae89e3226e57dcbf31c08f4d9e5743f32e193147a22/luma.core-1.12.0-py2.py3-none-any.whl
Collecting spidev; platform_system == "Linux" (from luma.core>=1.12.0->luma.oled)
  Using cached https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-q86lxe3n/spidev/`

I suppose, this is probably just a minor flaw, but as I'm more a user than a developer, that's where I already get stuck.

### Type of Raspberry Pi
RPi2B

#### Linux Kernel version
Linux raspy-mpd 4.0.7-v7+ #801 SMP PREEMPT Tue Jun 30 18:38:23 BST 2015 armv7l GNU/Linux

#### Expected behaviour
... successful install of luma.oled

#### Actual behaviour
`sudo -H pip3 install --upgrade luma.oled
Collecting luma.oled
  Using cached https://files.pythonhosted.org/packages/b0/fa/ef1065ad3644608d6127ce05da43b2b6ab1857b5b26983439f696bab6f4a/luma.oled-3.4.0-py2.py3-none-any.whl
Collecting luma.core>=1.12.0 (from luma.oled)
  Using cached https://files.pythonhosted.org/packages/a7/36/f10ac3202221c479bae89e3226e57dcbf31c08f4d9e5743f32e193147a22/luma.core-1.12.0-py2.py3-none-any.whl
Collecting spidev; platform_system == "Linux" (from luma.core>=1.12.0->luma.oled)
  Using cached https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
    ModuleNotFoundError: No module named 'setuptools'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-q86lxe3n/spidev/`

... any help is highly appreciated.
Tnx
bighegi.

rm-hull, 2020-03-08:
Try doing `sudo -H pip3 install setuptools` and then retry installing luma.oled again. I suspect you are running quite an old linux version based on the kernel version and date.

bighegi, 2020-03-09:
Thanks Richard, that does the trick :-).
If everything was that simple ...
Tnx
bighegi

closed 2020-03-09 by thijstriemstra